mavlink/MAVSDK #1866: battery information not correct with sitl

Amna-Smaoui, 2022-07-21:
Hi ! I'm using Mavsdk with python and I want get the battery.voltage and the battery.remaining_percent . At the begin it shows right value, but a few seconds later i don't get logic values and i don't know what's wrong . I'm working with SITL not a real drone.
`
[image]
[image]

julianoes, 2022-07-21:
I think that's just what SITL simulates, check the code:

https://github.com/PX4/PX4-Autopilot/blob/55563eba49f87624d7b73356ea3d421fee7ec081/src/modules/simulator/battery_simulator/BatterySimulator.cpp#L100

And there are the params:
https://docs.px4.io/main/en/advanced_config/parameter_reference.html#SIM_BAT_DRAIN
https://docs.px4.io/main/en/advanced_config/parameter_reference.html#SIM_BAT_MIN_PCT

Amna-Smaoui, 2022-07-23:
thank you for the links . the problem that battery.voltage does not exceed a certain threshold while the drone is on mission and logically it must decrease again . And the battery remaining it does not reflect the mission I created wher the drone makes several turns .
I'm using them to campare two missions where the number of turns is higher in one than the other .

julianoes, 2022-07-26:
Yes, that's a limitation of how PX4 simulates the battery, as I linked to above. I don't think there is much that MAVSDK can do about it. It's only receiving that data.

closed 2022-07-26 by julianoes