mavlink/MAVSDK #1867: Running example takeoff_and_land compiles errors on Macos

ayomidebajo, 2022-07-21:
After compiling the SDK and running the px4 simulator, I get this error when I run the first example (i.e takeoff_and_land)

This is the error below.
[image]

Note that I can control the drone from px4 using `commander takeoff` and `commander land`

Please is there anything I did wrong, I'm using MACOS Montenery by the way.

ayomidebajo, 2022-07-21:
The first output was different because I tried fixing the error, but it compiled to another error, so I reversed what I fixed. I don't have any experience writing c++😅

JonasVautherin, 2022-07-22:
Feels like you are using a new example with an older version of MAVSDK. Please use the examples from the tag that corresponds to your MAVSDK version (if you use `main`, take the examples from `main`, if you use `v1.4.4`, take the examples from tag `v1.4.4`).

ayomidebajo, 2022-07-26:
I downloaded the `v1.4.4` and it gave me the same error. I checked the docs. I use an M1 chip and the SDK doesn't really support M1 chip users for now 🥲

JonasVautherin, 2022-07-26:
> I downloaded the v1.4.4 and it gave me the same error.

What did you download? A package?

> I use an M1 chip and the SDK doesn't really support M1 chip users for now smiling_face_with_tear

If you build from sources, I don't see why it would not work.

Could you explain what steps you are following?

ayomidebajo, 2022-07-26:
I downloaded the package from `brew`. Then I cloned this repo.

ayomidebajo, 2022-07-26:
> > I downloaded the v1.4.4 and it gave me the same error.
> 
> What did you download? A package?
> 
> > I use an M1 chip and the SDK doesn't really support M1 chip users for now smiling_face_with_tear
> 
> If you build from sources, I don't see why it would not work.
> 
> Could you explain what steps you are following?

I downloaded the package from brew. Then I cloned this repo.

JonasVautherin, 2022-07-26:
Yeah that may not work for M1. Try to build MAVSDK from sources instead :+1:

ayomidebajo, 2022-07-26:
what package did you use to extract it?  I'm having issues using `dpkg`

JonasVautherin, 2022-07-26:
Sorry I did not get that :confused:

ayomidebajo, 2022-07-26:
Wrong question, sorry.

I tried building it, I just remembered that I did this method before and I kept having issues with `openssl` even after installing it and deleting it several times.

This is the error, I got when trying to build the sdk.
[image]

ayomidebajo, 2022-07-26:
My current version of `openssl` and the location it is installed.

[image]

JonasVautherin, 2022-07-26:
My guess is that you are lacking the SDKROOT, as defined e.g. here: https://github.com/mavlink/MAVSDK/blob/main/.github/workflows/main.yml#L514

julianoes, 2022-07-28:
https://stackoverflow.com/questions/51761599/cannot-find-stdio-h

julianoes, 2024-05-27:
:man_shrugging:

closed 2024-05-27 by julianoes