mavlink/MAVSDK #1872: set_acceleration_ned problem along down direction

PeymanAmirii, 2022-07-28:
When we use set_acceleration_ned(AccelerationNed(0, 0, 0)), the drones descend so fast and do not stay at a point. If we use set_acceleration_ned(AccelerationNed(0, 0, -0.1)), the droens relatively stay at a point. What is the problem?

julianoes, 2022-07-29:
First of all, trying to understand, what are you trying to do? What is the context?

I would assume that when you control acceleration, you need to "close the loop" outside, so you need to have a controller that controls acceleration based on a velocity or position setpoint that you are trying to reach.

However, this is not trivial, so the question is why you are trying to use the acceleration interface in the first place.

PeymanAmirii, 2022-07-29:
Thanks for your answer. We have used set_velocity_ned(VelocityNed(V_north, V_east, V_down)) to send the target velocity obtained based on P controller to get the quadcopter to a set point. Now, we want to use set_acceleration_ned(AccelerationNed(A_north, A_east, A_down)) to send the target acceleration obtained based on P controller to get the quadcopter to a set point. 
When we use this command set_velocity_ned(VelocityNed(0, 0, 0)), the quadcopter stays at a point but when we use this cammnd set_acceleration_ned(AccelerationNed(0, 0, 0)), the quadcopter descends.

julianoes, 2022-08-02:
> now, we want to use the command set_acceleration_ned

Ok, why? What are you trying to solve or improve?

PeymanAmirii, 2022-08-02:
Thanks for your answer. We want to see if a P controller with acceleration command can be more accurate than a P controller with velocity command or not. Is it possible to find out why when use the command set_acceleration_ned(AccelerationNed(0, 0, 0)), the drones descent but when we use the command set_velocity_ned(VelocityNed(0, 0, 0)), they stay at a point?
Thanks.

julianoes, 2022-08-02:
If you command set_velocity_ned(VelocityNed(0, 0, 0)) I expect it to drift in position over time, unless you are closing the loop with a position P controller around it.

In the same way if you command set_acceleration_ned(AccelerationNed(0, 0, 0)) I expect it to drift in velocity over time, unless you are closing the loop with a velocity (and eventually position) controller around it.

I assume it works much better with velocity inputs because you only integrate drift once, so the drift is slow an manageable. If you control acceleration however, you are integrating twice, so acceleration -> velocity -> position, so your drift accumulates much quicker.

PeymanAmirii, 2022-08-05:
Thanks fo your detailed answer. As the last question, how acceleration is measured in this case? By derivating the position obtained using GPS, twice or using accelerometers of IMU or a fusion of these two methods?

julianoes, 2022-08-05:
It's estimated using the ekf2 estimator. So it contains IMU, GPS, whatever is available.