mavlink/MAVSDK #1873: Timeout errors when downloading mission to autopilot

pbuschmeyer, 2022-07-30:
We are downloading missions to the autopilot from a companion computer (Jetson Nano).  The communication is via autopilot TELEM2 port configured at 115,200, "Onboard", 80000 b/s, forwarding enabled, to a serial port on the companion computer.  We have mavlink-router running in the companion computer linking the serial port on the companion computer to UDP.  Our C++ application connects to mavlink-router via UDP with MAVSDK.  So,

autopilot <-> serial <-> mavlink-router <-> UDP <-> MAVSDK <-> custom application.

We are trying to download missions to the autopilot but they fail with timeout errors.  We are waiting for the system status to be OK before downloading.  We are in the "HOLD" flight mode and "OnGround" landed state.  The attached text file contains a log of the error.

[MAVSDK Mission Download Log 2022-07-29.txt](https://github.com/mavlink/MAVSDK/files/9224167/MAVSDK.Mission.Download.Log.2022-07-29.txt)

We basically used the MAVSDK Mission example code as shown here.

`		case MSG_MAV_MISSION_TO_AP:
		{
			auto mavData = std::static_pointer_cast<MAVData>(msg->aParam);
			string strFQPNFilename = mavData->m_strData;
//#ifdef _DEBUG
//			printf("MavThread::MSG_MAV_MISSION_TO_AP Request received to upload mission %s to Autopilot\n", 
//				strFQPNFilename.c_str());
//#endif

			std::pair<MissionRaw::Result, MissionRaw::MissionImportData> import_res = mission_raw.import_qgroundcontrol_mission(strFQPNFilename);
			string strMissionResult = "MavThread::MSG_MAV_MISSION_TO_AP imported mission " + strFQPNFilename + " from QGroundControl with result: " + PrintMissionRawResult(import_res.first);
			std::shared_ptr<MAVData> spsDownloadMissionStatus(new MAVData(import_res.first == MissionRaw::Result::Success, (int)import_res.first, strMissionResult));
			m_pCallingThread->PostThreadMsg(MSG_MAV_MISSION_IMPORT, spsDownloadMissionStatus, 0);
//#ifdef _DEBUG
//			printf("MavThread::MSG_MAV_MISSION_TO_AP import mission %s from QGroundControl %s\n", strFQPNFilename.c_str(), import_res.first == MissionRaw::Result::Success ? "successfully" : "failed");
//#endif
			if (import_res.first == MissionRaw::Result::Success)
			{
				string strMissionResult = "MavThread::MSG_MAV_MISSION_TO_AP mission " + strFQPNFilename + " non-zero items: " + PrintMissionRawResult(import_res.first);
				std::shared_ptr<MAVData> spsDownloadMissionStatus(new MAVData(import_res.second.mission_items.size() > 0, (int)import_res.first, strMissionResult));
				m_pCallingThread->PostThreadMsg(MSG_MAV_MISSION_ITEMCOUNT, spsDownloadMissionStatus, 0);
				if (import_res.second.mission_items.size() > 0)
				{
					const MissionRaw::Result upload_result = mission_raw.upload_mission(import_res.second.mission_items);
					string strMissionResult = "MavThread::MSG_MAV_MISSION_TO_AP upload mission " + strFQPNFilename + " with result: " + PrintMissionRawResult(upload_result);
					std::shared_ptr<MAVData> spsDownloadMissionStatus(new MAVData(upload_result == MissionRaw::Result::Success, (int)import_res.first, strMissionResult));
					m_pCallingThread->PostThreadMsg(MSG_MAV_MISSION_DOWNLOAD, spsDownloadMissionStatus, 0);
//#ifdef _DEBUG
//					printf("MavThread::MSG_MAV_MISSION_TO_AP Upload mission %s to Autopilot %s\n", strFQPNFilename.c_str(), upload_result == MissionRaw::Result::Success ? "successfully" : "failed");
//#endif
				}
			}
		}
			break;
`

That said, this works fine in the HITL mode, and works "mostly" if we power the autopilot through the USB cable (QGroundControl is NOT running).  When we run this using just the battery to power everything, the mission download times out EVERY TIME.

Questions:  Are there conditions under which the autopilot will not accept the download of a mission? Must the download be accomplished with a USB cable only (hence it "works" when we power via USB)?
Do we have the communications balled up some how?

Any help you can provide will be GREATLY appreciated!!
Paul

pbuschmeyer, 2022-07-31:
SOLVED!!!!!  Serial port baud rate was too slow to mavlink-router.

closed 2022-07-31 by pbuschmeyer

julianoes, 2022-08-02:
I see, thanks for the note!