mavlink/MAVSDK #1875: follow_me.cpp won't build

m-yoshida04, 2022-08-01:
I got the MAVSDK source codes using git command of `git clone https://github.com/mavlink/MAVSDK.git --recursive`.
Then I tried to build them per the following commands in the MAVSDK/examples/follow_me.
```
cmake -Bbuild -H.
cmake --build build -j4`
```

However, following message came in and I can not never built.

```
[ 33%] Building CXX object CMakeFiles/follow_me.dir/fake_location_provider.cpp.o
[ 66%] Building CXX object CMakeFiles/follow_me.dir/follow_me.cpp.o
/home/uav-px4/MAVSDK/examples/follow_me/follow_me.cpp: In function ‘std::shared_ptr<mavsdk::System> get_system(mavsdk::Mavsdk&)’:
/home/uav-px4/MAVSDK/examples/follow_me/follow_me.cpp:41:13: error: ‘NewSystemHandle’ is not a member of ‘mavsdk::Mavsdk’
   41 |     Mavsdk::NewSystemHandle handle = mavsdk.subscribe_on_new_system([&mavsdk, &prom, &handle]() {
      |             ^~~~~~~~~~~~~~~
/home/uav-px4/MAVSDK/examples/follow_me/follow_me.cpp:51:6: error: expected primary-expression before ‘)’ token
   51 |     });
      |      ^
/home/uav-px4/MAVSDK/examples/follow_me/follow_me.cpp: In function ‘int main(int, char**)’:
/home/uav-px4/MAVSDK/examples/follow_me/follow_me.cpp:115:16: error: ‘FlightModeHandle’ is not a member of ‘mavsdk::Telemetry’
  115 |     Telemetry::FlightModeHandle handle =
      |                ^~~~~~~~~~~~~~~~
/home/uav-px4/MAVSDK/examples/follow_me/follow_me.cpp:121:10: error: expected primary-expression before ‘)’ token
  121 |         });
      |          ^
/home/uav-px4/MAVSDK/examples/follow_me/follow_me.cpp:135:12: error: ‘struct mavsdk::FollowMe::Config’ has no member named ‘follow_height_m’
  135 |     config.follow_height_m = 10.0;
      |            ^~~~~~~~~~~~~~~
/home/uav-px4/MAVSDK/examples/follow_me/follow_me.cpp:136:12: error: ‘struct mavsdk::FollowMe::Config’ has no member named ‘follow_angle_deg’
  136 |     config.follow_angle_deg = 180.0f;
      |            ^~~~~~~~~~~~~~~~
/home/uav-px4/MAVSDK/examples/follow_me/follow_me.cpp:173:15: error: ‘class mavsdk::Telemetry’ has no member named ‘unsubscribe_flight_mode’; did you mean ‘subscribe_flight_mode’?
  173 |     telemetry.unsubscribe_flight_mode(handle);
      |               ^~~~~~~~~~~~~~~~~~~~~~~
      |               subscribe_flight_mode
/home/uav-px4/MAVSDK/examples/follow_me/follow_me.cpp:173:39: error: ‘handle’ was not declared in this scope
  173 |     telemetry.unsubscribe_flight_mode(handle);
      |                                       ^~~~~~
make[2]: *** [CMakeFiles/follow_me.dir/build.make:63: CMakeFiles/follow_me.dir/follow_me.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/follow_me.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

```
It can not find NewSystemHandle in the mavsdk.h.

Does anybody can help me to build?

Thanks.

m-yoshida04, 2022-08-01:
The build is successfully completed when it is done from the sources. 
(Sources are downloaded by entering the command of ```git clone https://github.com/mavlink/MAVSDK.git```and build according to ```https://mavsdk.mavlink.io/main/en/cpp/guide/build.html```)

However, the vehicle will not respond to the command even after the follow me example is run by the command of ``` build/follow_me udp://:14540 ```.
In other examples such as ```fly_mission``` and ```takeoff_and_land```, the vehicle responds to the command as expected.

Does anybody know how I can solve this issue?
I have the latest PX4 source codes by entering the commands of  ``` git clone https://github.com/PX4/PX4-Autopilot.git --recursive``` and uses QGC daily build as GCS.

Thank you for your help in advance.

julianoes, 2022-08-01:
You are trying to build the example from the mavsdk `main` branch which requires the library to be updated to main as well. The `main` branch has quite a few API changes and will be released as version 2 soon.

Until then, I recommend checking out the branch `v1.4` (or the release tag) which matches what you have installed.

m-yoshida04, 2022-08-01:
Thank you for your quick reply. I would like to run the example for new follow me mode in MAVSDK. Is branch ```v1.4``` is good one to be checked out and run?

julianoes, 2022-08-02:
That's v2.0 then as it has changed recently. In that case you need to use the new subscription API. Check the changes here:
https://github.com/mavlink/MAVSDK/pull/1768

m-yoshida04, 2022-08-02:
Thank you again for your kind reply. 
It seems *fake_location_provider.cpp* in mavsdk ```main``` branch does provide target lateral and longitudinal information but does not provide target altitude. If it is correct, I believe this is the reason why the vehicle does not respond to the follow me command via MAVSDK in my simulation while the follow me example is executed.  The outputs of *pos_est* and *vel_est* in *follow_target_estimator* message are always 'NAN' in my simulation.

m-yoshida04, 2022-08-02:
I did succeed to run the example for new follow me mode via MAVSDK. So, this can be closed. Thank you.

julianoes, 2022-08-02:
Oh nice! Thanks for the note.

closed 2022-08-02 by julianoes