mavlink/MAVSDK #1879: MAVSDK v1.4.4 with PX4 v1.13.0

weagle08, 2022-08-05:
I had no problems connecting MAVSDK to PX4 v1.12.3, but it does not seem to connect to the latest v1.13.0. Any thoughts on what could cause this? I'm using just the core library and plugins with C++.

Thanks

julianoes, 2022-08-05:
This might be because PX4 no longer runs mavlink on USB by default but only after you send it heartbeats.

Try opening the question with the option to send heartbeats.
https://github.com/mavlink/MAVSDK/blob/425577aa5629d591a854a65fd1f2dd6d88b526d9/src/mavsdk/core/include/mavsdk/mavsdk.h#L199-L200

weagle08, 2022-08-05:
That works! Thanks for the quick response!
```c++
mavsdk::Mavsdk::Configuration sdkConfig(mavsdk::Mavsdk::Configuration::UsageType::CompanionComputer);
sdkConfig.set_always_send_heartbeats(true);
sdkConfig.set_system_id(this->_fcSystem->get_system_id());
sdkConfig.set_component_id(this->_fcSystem->component_ids().front());
this->_msdk->set_configuration(sdkConfig);
```

closed 2022-08-05 by weagle08

julianoes, 2022-08-08:
Nice, thanks for the feedback. We should probably add a note to the docs about this:
https://github.com/mavlink/MAVSDK-docs/blob/main/en/cpp/guide/connections.md