mavlink/MAVSDK #1881: PX4/MAVSDK failsafe test

Rubenapc, 2022-08-17:
I have a mission for a drone flight in MAVSDK but to test possible system failures during the mission in the PX4 terminal I do ("failure gps off", for example) and the drone starts to land where it is. I wanted to know if I can write something in the code for the drone to do return to launch instead of land because I have some Rally Points defined and so when testing the failure it landed on one of the rally points.

julianoes, 2022-08-18:
So you want to land on the rally points when you lose GPS? The problem I see is that you don't know how to get to the rally points without GPS, so you really can't do much else than land.

This is unless you have some other mean of positioning, like flow or VIO.

Rubenapc, 2022-08-19:
Really without GPS it makes sense that the drone only has the ability to land by lowering the height in the position it is in. So, in case I lose the GPS, I have the possibility of just landing because "flow" or "VIO" will not be my objective.
Using the same situation for another sensor failure, in the MAVSDK class failure there are other sensor and system options. I have the option of injecting a failure into the code or writing it manually in the PX4 terminal ("failure baro off") and this is the one I prefer to use because then I can get the failure only if I want to. My problem is how the code detects that there was a certain failure inserted in the terminal and if the drone action is defined by me, for example "Return to Launch" and so it lands at Rally Points.

julianoes, 2022-08-21:
Yer, that's not all straight forward. Best is probably to check the PX4 docs about failsafes, and try out things in SITL.

Also note that not all failures are supported in PX4 SITL and even less in "real" PX4.