mavlink/MAVSDK #1890: How can i use the function in the Telemetry::Battery

scps990210, 2022-09-02:
How can i use the function   float mavsdk::Telemetry::Battery::remaining_percent { float(NAN)}?
I declare auto telemetry = Telemetry{system}; and then use telemetry.remaining_percent but it doesn't work.
Can someone help me? thanks!

JonasVautherin, 2022-09-02:
Did you try the [battery](https://github.com/mavlink/MAVSDK/tree/main/examples/battery) example? Can you show us your code? What autopilot are you using (PX4/Ardupilot)? Do the examples generally work?

scps990210, 2022-09-02:
I haven't try the battery example but i saw the function in this [page](https://mavsdk.mavlink.io/main/en/cpp/api_reference/structmavsdk_1_1_telemetry_1_1_battery.html) and considering how to use it 
I'm trying to write a discriminant that consider whether the battery remaining percnet is lower than 20% or not so my code is 
`auto telemetry = Telemetry{system};
while (telemetry.remaining_percent>0.2) {
        //do something;
    }
//do RTL `

julianoes, 2022-09-02:
This logic should be done the autopilot. For PX4, check these parameters:

https://docs.px4.io/main/en/advanced_config/parameter_reference.html#BAT_CRIT_THR

https://docs.px4.io/main/en/advanced_config/parameter_reference.html#COM_LOW_BAT_ACT

scps990210, 2022-09-04:
Can you show me how to use these two functions? thanks!

julianoes, 2022-09-05:
You are supposed to set these parameters in PX4, so e.g. by using QGroundControl, or in the Shell, `param set BAT_CRIT_THR 0.2`...

closed 2024-05-27 by julianoes