mavlink/MAVSDK #1891: in MAVSDK/examples/fly_mission/fly_mission.cpp

scps990210, 2022-09-02:
I have some question about this program as follows, hope you can solve my doubts, thank you!
1.Why in line 214-216 we need to set 
while (!want_to_pause) {
    sleep_for(seconds(1));
}
If the previous work in line 201-205 didn't change the want_to_pause to true, will the program block in line 214-216?
2.Does the pause mission behavior in 218-224 is indispensable or just an example of pausing it?
3.If i only have 1 mission, can i just write 
Mission::Result start_mission_result = mission.start_mission();
    if (start_mission_result != Mission::Result::Success) {
        std::cerr << "Starting mission failed: " << start_mission_result << '\n';
        return 1;
    }
        std::cout<<"Mission completed"<< '\n';
remove line 195-206 and 214-238

julianoes, 2022-09-02:
The pause mission logic is just an example of how you can pause it, if you want.

scps990210, 2022-09-04:
Got it! Thank you!

closed 2022-11-22 by julianoes