mavlink/MAVSDK #1899: [Discussion] How to set a parameter and reboot the vehicle for PX4 MAVSDK SITL Test

junwoo091400, 2022-09-16:
## What I want to do
I wanted to have the following feature on [MAVSDK Test](https://github.com/PX4/PX4-Autopilot/tree/main/test/mavsdk_tests) in PX4 SITL:
1. Set the parameter in PX4 (to set it to a non-default value)
2. Reboot the vehicle (Or start the vehicle directly with the modified parameter from the beginning)
3. Perform a SITL MAVSDK test with the vehicle that had the modified parameter when it booted up

The reason for this was for the [Payload delivery SITL test PR](https://github.com/PX4/PX4-Autopilot/pull/20230), I needed a way to set the parameter and reboot the vehicle, in order to have the startup script recognize the non-default parameter to enable the `payload_deliverer` module, and start the module, which is necessary for payload delivery feature.

## Problem
Currently it seems like we can't arbitrarily set the parameter before booting up, neither can we reboot the vehicle in the SITL test function.

### Proposed Solutions
1. Have a way to reboot the vehicle after setting the parameter in the mavsdk sitl test pipeline
2. Have a way to load an arbitrary / modified parameter through MAVSDK interface for the drone

## Discussion 
If we want to support such feature, what would be the best way to implement it? Are there any other solutions? Would this be possible without modifying MAVSDK?

JonasVautherin, 2022-09-16:
So the goal of the test is to check that the parameter is set correctly after reboot, but SITL resets the params when it boots, right?

I don't really see how this is related to MAVSDK, though :thinking:. It feels like you need SITL to persist the params (instead of resetting after reboot), and that's not how SITL does it, right?

julianoes, 2022-09-18:
1. From what I know there is not yet a way to reboot/restart PX4 from within PX4 in SITL but only externally via helper scripts. So for the PX4 mavsdk_tests you would have to script it there.
2. You can always load/get an arbitrary param from PX4. I think that part is not a problem.

JonasVautherin, 2022-09-18:
But then if it requires custom code to make it work in SITL, is it really a test? It feels like params persistence after reboot is just not a SITL feature. We can simulate it with external scripts, and then we can test that those external scripts simulate it correctly, but at the end of the day it's not testing the actual feature :sweat_smile:

closed 2024-05-27 by julianoes