mavlink/MAVSDK #1900: Can't find CURLConfig.cmake for mavsdk-windows-x64-release

krr27, 2022-09-17:
I'm trying to install MAVSDK library on Windows x64
I'm following the instructions under MAVSDK QuickStart (https://mavsdk.mavlink.io/main/en/cpp/quickstart.html)
I'm currently stuck on the "Build and Try Example" step

I entered the following (in a Command Prompt):
D:\MAVSDK\examples\takeoff_and_land>cmake -Bbuild -DCMAKE_PREFIX_PATH=d:/mavsdk-windows-x64-release -DCURL_DIR=D:\vcpkg\installed\x86-windows\share\curl -DCMAKE_TOOLCHAIN_FILE=D:/vcpkg/scripts/buildsystems/vcpkg.cmake -H.

I get the following output:
-- Selecting Windows SDK version 10.0.22621.0 to target Windows 10.0.22000.
CMake Error at C:/Program Files/CMake/share/cmake-3.23/Modules/CMakeFindDependencyMacro.cmake:47 (find_package):
  Could not find a configuration file for package "CURL" that is compatible
  with requested version "".

  The following configuration files were considered but not accepted:

    D:/vcpkg/installed/x86-windows/share/curl/CURLConfig.cmake, version: 7.84.0-DEV (32bit)

Call Stack (most recent call first):
  D:/mavsdk-windows-x64-release/lib/cmake/MAVSDK/MAVSDKConfig.cmake:7 (find_dependency)
  CMakeLists.txt:12 (find_package)

Note that I set the following in the command, because earlier attempts indicated that it needs CURLConfig.cmake, and that's the only directory where I found it
-DCURL_DIR=D:\vcpkg\installed\x86-windows\share\curl

I'm guessing that I need CURLConfig.cmake (or a substitute) under D:\vcpkg\installed\x64-windows\share, but I'm not sure and don't know how to get or generate it.

Any suggestions for what to try next?

JonasVautherin, 2022-09-17:
It's weird because the [CMake docs](https://cmake.org/cmake/help/latest/command/find_package.html#config-mode-version-selection) say:

> When the [version] argument is given, Config mode will only find a version of the package that claims compatibility with the requested version (see [format specification](https://cmake.org/cmake/help/latest/command/find_package.html#find-package-version-format)).

So I would expect that `""` means that the `[version]` argument was not given :thinking:.

Anyway can you try to find the [`MAVSDKConfig.cmake`](https://github.com/mavlink/MAVSDK/blob/main/src/MAVSDKConfig.cmake.in#L7) file in your installation and set a version there (I would try with "7"), to see if that helps?

So something like (assuming you have a CMake version >= 3.9.0):

```
find_dependency(CURL 7 REQUIRED CONFIG)
```

krr27, 2022-09-18:
I tried your suggestion (find_dependency(CURL 7 REQUIRED CONFIG)). The resulting cmake output was the same, except for the following:

  Could not find a configuration file for package "CURL" that is compatible
  with requested version "7".

I'm new to the cmake business (and MAVSDK), but my best guess is that I need CURLConfig.cmake (or a substitute) for x64-windows (e.g., under D:\vcpkg\installed\x64-windows\share), since I'm using mavsdk-windows-x64-release. The cmake output says that the CURLConfig.cmake I'm using is "version: 7.84.0-DEV (32bit)".

Comments at the top of CURLConfig.cmake indicate that it was autogenerated from curl-config.cmake.in. I have a version of curl-config.cmake.in (just one, which is in D:\curl\CMake). Did I miss (or screw up) an earlier step that would have generated an x64 (or otherwise compatible) version of CURLConfig.cmake? I'll review the installation steps again.

Thanks for the suggestion. I think I'm a bit closer now. I'm going to continue reading the docs and researching, but meanwhile I would appreciate any additional suggestions.

JonasVautherin, 2022-09-18:
Ok, so now your error says (I'm guessing since you did not copy it entirely):

```
Could not find a configuration file for package "CURL" that is compatible
with requested version "7".

The following configuration files were considered but not accepted:

D:/vcpkg/installed/x86-windows/share/curl/CURLConfig.cmake, version: 7.84.0-DEV (32bit)
```

That's promising, I believe. It now requests `7` and finds `7.84.0-DEV`. Did you try other values than `7`, like for instance `7.84.0` and `7.84.0-DEV`?  Does it complain all the same?

krr27, 2022-09-18:
Below are the results of setting CURL 7.84.0 and CURL 7.84.0-DEV

*** Using find_dependency(CURL 7.84.0 REQUIRED CONFIG):

D:\MAVSDK\examples\takeoff_and_land>cmake -Bbuild -DCMAKE_PREFIX_PATH=d:/mavsdk-windows-x64-release -DCURL_DIR=D:\vcpkg\packages\curl_x86-windows\share\curl -DCMAKE_TOOLCHAIN_FILE=D:/vcpkg/scripts/buildsystems/vcpkg.cmake -H.
-- Selecting Windows SDK version 10.0.22621.0 to target Windows 10.0.22000.
CMake Error at D:/vcpkg/scripts/buildsystems/vcpkg.cmake:829 (_find_package):
  Could not find a configuration file for package "CURL" that is compatible
  with requested version "7.84.0".

  The following configuration files were considered but not accepted:

    D:/vcpkg/packages/curl_x86-windows/share/curl/CURLConfig.cmake, version: 7.84.0-DEV (32bit)

Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.23/Modules/CMakeFindDependencyMacro.cmake:47 (find_package)
  D:/mavsdk-windows-x64-release/lib/cmake/MAVSDK/MAVSDKConfig.cmake:7 (find_dependency)
  D:/vcpkg/scripts/buildsystems/vcpkg.cmake:829 (_find_package)
  CMakeLists.txt:12 (find_package)

-- Configuring incomplete, errors occurred!
See also "D:/MAVSDK/examples/takeoff_and_land/build/CMakeFiles/CMakeOutput.log".

*** Using find_dependency(CURL 7.84.0-DEV REQUIRED CONFIG):

D:\MAVSDK\examples\takeoff_and_land>cmake -Bbuild -DCMAKE_PREFIX_PATH=d:/mavsdk-windows-x64-release -DCURL_DIR=D:\vcpkg\packages\curl_x86-windows\share\curl -DCMAKE_TOOLCHAIN_FILE=D:/vcpkg/scripts/buildsystems/vcpkg.cmake -H.
-- Selecting Windows SDK version 10.0.22621.0 to target Windows 10.0.22000.
CMake Error at D:/vcpkg/scripts/buildsystems/vcpkg.cmake:829 (_find_package):
  _find_package called with invalid argument "7.84.0-DEV"
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.23/Modules/CMakeFindDependencyMacro.cmake:47 (find_package)
  D:/mavsdk-windows-x64-release/lib/cmake/MAVSDK/MAVSDKConfig.cmake:7 (find_dependency)
  D:/vcpkg/scripts/buildsystems/vcpkg.cmake:829 (_find_package)
  CMakeLists.txt:12 (find_package)

CMake Error at D:/vcpkg/scripts/buildsystems/vcpkg.cmake:829 (_find_package):
  Found package configuration file:

    D:/mavsdk-windows-x64-release/lib/cmake/MAVSDK/MAVSDKConfig.cmake

  but it set MAVSDK_FOUND to FALSE so package "MAVSDK" is considered to be
  NOT FOUND.  Reason given by package:

  MAVSDK could not be found because dependency CURL could not be found.

Call Stack (most recent call first):
  CMakeLists.txt:12 (find_package)

-- Configuring incomplete, errors occurred!
See also "D:/MAVSDK/examples/takeoff_and_land/build/CMakeFiles/CMakeOutput.log".

krr27, 2022-09-18:
I downloaded mavsdk-windows-x64-release.zip (from https://github.com/mavlink/MAVSDK/releases) and unzipped it to get d:/mavsdk-windows-x64-release

I don't recall compiling MAVSDK (e.g., from D:\MAVSDK root). I think I have only attempted to build D:\MAVSDK\examples\takeoff_and_land

JonasVautherin, 2022-09-18:
Ok, it seems like `x64` comes from [here](https://github.com/mavlink/MAVSDK/blob/main/.github/workflows/main.yml#L633), which seems like a typo. Then we call it `win32` [here](https://github.com/mavlink/MAVSDK/blob/main/.github/workflows/main.yml#L649), but honestly I don't really know how to tell if that's correct or not.

Would you happen to know how to check if the windows library you installed is 32 or 64bits?

Also I find this weird:

```
[...] with requested version "7.84.0".

The following configuration files were considered but not accepted:

D:/vcpkg/packages/curl_x86-windows/share/curl/CURLConfig.cmake, version: 7.84.0-DEV (32bit)
```

Can you try to fetch curl in 64bits with vcpkg and give it a try?

krr27, 2022-09-19:
Sorry, I don't know how to check whether the windows library I installed is 32 or 64 bits.
How would I do that?
Are you referring to the MAVSDK library (in D:\mavsdk-windows-x64-release)?
Also, how would I fetch curl in 64 bits and try it?
Sorry, but I'm new to cmake, curl, and vcpkg.
At this point I'm just following the MAVSDK installation instructions and hoping it works, so I can start using it, because it sounds great!

krr27, 2022-09-19:
Below is the complete output that I get, after deleting the build directory and rerunning the cmake command.

The 1st 14 lines (the ones starting with --) appear to indicate that it got pretty far before getting an error.

Also, I don't see any indication of an error in the attached 
[CMakeOutput.zip](https://github.com/mavlink/MAVSDK/files/9601732/CMakeOutput.zip)
CMakeOutput.log that it generated.

_D:\MAVSDK\examples\takeoff_and_land>cmake -Bbuild -DCMAKE_PREFIX_PATH=d:/mavsdk-windows-x64-release -DCURL_DIR=D:\vcpkg\packages\curl_x86-windows\share\curl -DCMAKE_TOOLCHAIN_FILE=D:/vcpkg/scripts/buildsystems/vcpkg.cmake -H.
-- Building for: Visual Studio 17 2022
-- Selecting Windows SDK version 10.0.22621.0 to target Windows 10.0.22000.
-- The C compiler identification is MSVC 19.32.31329.0
-- The CXX compiler identification is MSVC 19.32.31329.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Professional/VC/Tools/MSVC/14.32.31326/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Professional/VC/Tools/MSVC/14.32.31326/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at D:/vcpkg/scripts/buildsystems/vcpkg.cmake:829 (_find_package):
  Could not find a configuration file for package "CURL" that is compatible
  with requested version "7.84.0".

  The following configuration files were considered but not accepted:

    D:/vcpkg/packages/curl_x86-windows/share/curl/CURLConfig.cmake, version: 7.84.0-DEV (32bit)

Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.23/Modules/CMakeFindDependencyMacro.cmake:47 (find_package)
  D:/mavsdk-windows-x64-release/lib/cmake/MAVSDK/MAVSDKConfig.cmake:7 (find_dependency)
  D:/vcpkg/scripts/buildsystems/vcpkg.cmake:829 (_find_package)
  CMakeLists.txt:12 (find_package)

-- Configuring incomplete, errors occurred!
See also "D:/MAVSDK/examples/takeoff_and_land/build/CMakeFiles/CMakeOutput.log"._

JonasVautherin, 2022-09-19:
Would you mind formatting your answers, so that I don't have to do it myself to make it readable?

> Below is the complete output that I get

Is that different from what you had below? I don't see a difference, did I miss something? :thinking: 

> Also, how would I fetch curl in 64 bits and try it? Sorry, but I'm new to cmake, curl, and vcpkg.

I would suggest reading a tutorial, I'm pretty sure that vcpkg has that somewhere :innocent:

krr27, 2022-09-19:
The only difference is that the output I put in the last message includes the output prior to the error message. I thought maybe that part of the output, which appears to indicate steps that succeeded, would provide additional clues. The error messages are the same as before.

I have been reading about cmake, curl, and vcpkg, so now I at least know what they are, but I don't know enough yet to efficiently track down the cause of the error.

I tried installing vcpkg curl version 7.82.0, which is the only vcpkg curl version I see listed on (https://vcpkg.io/en/packages.html), and which that website indicates is compatible with x64-windows (and x86-windows). The output from trying to install it is pasted below. It appears to me that what I have been using is a slightly newer version of vcpkg curl, which I would expect to also be compatible with x64-windows.

    D:\vcpkg>.\vcpkg install curl
    Computing installation plan...
    The following packages are already installed:
        curl[core,non-http,schannel,ssl,sspi,tool]:x86-windows -> 7.84.0#2
    curl:x86-windows is already installed
    Restored 0 package(s) from C:\Users\krr\AppData\Local\vcpkg\archives in 86.4 us. Use --debug to see more details.**
    
    Total elapsed time: 25.3 ms
    
    curl provides CMake targets:

    # this is heuristically generated, and may not be correct
    find_package(CURL CONFIG REQUIRED)
    target_link_libraries(main PRIVATE CURL::libcurl)

JonasVautherin, 2022-09-19:
`.\vcpkg install curl:x64-windows`?

krr27, 2022-09-20:
That worked!!! Thanks!!!

It put a CURLConfig.cmake in D:\vcpkg\installed\x64-windows\share\curl, which cmake accepted.

I hit another snag (jsoncpp related), but I now have at least some idea what to look for and try, so I'll see how far I can get before asking for help.

krr27, 2022-09-20:
The cmake configuration step now appears to succeed, but the build step (cmake --build build -j8) fails (with link errors).

I have spent a couple of hours trying to track down the cause, but I haven't made any progress. I expect that the error messages might mean a lot more to someone familiar with the code.

Should I close this topic and start a new one, or just continue with this one?

JonasVautherin, 2022-09-20:
Let's start with a new one, I would say :+1:

closed 2022-09-20 by JonasVautherin