mavlink/MAVSDK #1902: Errors building example/takeoff_and_land, following initial MAVSDK installation/configuration

krr27, 2022-09-20:
I'm attempting to use the following command to build example/takeoff_and_land, following initial MAVSDK installation/configuration:

>D:\MAVSDK\examples\takeoff_and_land>cmake --build build -j8

System Info:
- Windows 11 Pro
- x64-based PC
- Visual Studio 2022

I'm using mavsdk-windows-x64-release.zip (prebuilt library and headers), but it appears that it's trying to build a Debug version, and I haven't been able to determine how to change that.

Output of the build command:

    D:\MAVSDK\examples\takeoff_and_land>cmake --build build -j8
    Microsoft (R) Build Engine version 17.2.1+52cd2da31 for .NET Framework
    Copyright (C) Microsoft Corporation. All rights reserved.
    
      Checking Build System
      Building Custom Rule D:/MAVSDK/examples/takeoff_and_land/CMakeLists.txt
      takeoff_and_land.cpp
    mavsdk.lib(system.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in     takeoff_and_land.obj [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
    mavsdk.lib(system.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in takeoff_and_land.obj [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
    mavsdk.lib(connection_result.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in takeoff_and_land.obj [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
    mavsdk.lib(connection_result.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in takeoff_and_land.obj [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
    mavsdk.lib(mavsdk.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in takeoff_and_land.obj [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
    mavsdk.lib(mavsdk.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in takeoff_and_land.obj [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
    mavsdk.lib(action.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in takeoff_and_land.obj [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
    mavsdk.lib(action.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in takeoff_and_land.obj [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
    mavsdk.lib(telemetry.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in takeoff_and_land.obj [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
    mavsdk.lib(telemetry.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in takeoff_and_land.obj [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
    mavsdk.lib(system_impl.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in takeoff_and_land.obj [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
    mavsdk.lib(system_impl.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in takeoff_and_land.obj [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
    mavsdk.lib(mavsdk_impl.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in takeoff_and_land.obj [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
    mavsdk.lib(mavsdk_impl.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in takeoff_and_land.obj [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
    mavsdk.lib(action_impl.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in takeoff_and_land.obj [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
    mavsdk.lib(action_impl.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in takeoff_and_land.obj [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
    mavsdk.lib(telemetry_impl.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in takeoff_and_land.obj [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
    mavsdk.lib(telemetry_impl.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in takeoff_and_land.obj [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
    mavsdk.lib(mavsdk_time.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in takeoff_and_land.obj [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
    mavsdk.lib(mavsdk_time.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in takeoff_and_land.obj [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
    mavsdk.lib(call_every_handler.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in takeoff_and_land.obj [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
    mavsdk.lib(call_every_handler.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in takeoff_and_land.obj [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
    mavsdk.lib(timeout_handler.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in takeoff_and_land.obj [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
    mavsdk.lib(timeout_handler.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in takeoff_and_land.obj [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
    mavsdk.lib(log.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in takeoff_and_land.obj [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
    mavsdk.lib(log.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in takeoff_and_land.obj [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
    mavsdk.lib(mavlink_parameters.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in takeoff_and_land.obj [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
    mavsdk.lib(mavlink_parameters.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in takeoff_and_land.obj [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
    mavsdk.lib(mavlink_command_receiver.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in takeoff_and_land.obj [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
    mavsdk.lib(mavlink_command_receiver.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in takeoff_and_land.obj [    D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
    mavsdk.lib(mavlink_command_sender.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in takeoff_and_land.obj [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
    mavsdk.lib(mavlink_command_sender.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in takeoff_and_land.obj [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
[... cut ...] k::PluginImplBase *>(class mavsdk::PluginImplBase * *,class ma
    vsdk::PluginImplBase * *,class mavsdk::PluginImplBase * const)" (??$__std_find_trivial@PEAVPluginImplBase@mavsdk@@PEAV12@@@YAPEAPEAVPluginImplBase@mavsdk@@PEAPEAV01@0QEAV01@@Z) [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
    D:\MAVSDK\examples\takeoff_and_land\build\Debug\takeoff_and_land.exe : fatal error LNK1120: 1 unresolved externals [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]

JonasVautherin, 2022-09-20:
> 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug'

That's your issue. Some conflict between _some_ libraries being built in Release mode and _some_ in Debug mode.

Can you try maybe those two?

```
cmake --build build --config Release
```

and

```
cmake --build build --config Debug
```

krr27, 2022-09-20:
cmake --build build --config Debug

Produced the same output as before, which seems to confirm that it built Debug config by default

cmake --build build --config Release

Produces the following output:

    D:\MAVSDK\examples\takeoff_and_land>cmake --build build --config Release
    Microsoft (R) Build Engine version 17.2.1+52cd2da31 for .NET Framework
    Copyright (C) Microsoft Corporation. All rights reserved.
    
    mavsdk.lib(system_impl.obj) : error LNK2019: unresolved external symbol __std_find_trivial_8 referenced in function "class     mavsdk::PluginImplBase * * __cdecl __std_find_trivial<class mavsdk::PluginImplBase *,class mavsdk::PluginImplBase *>(class mavsdk::PluginImplBase * *,class ma
    vsdk::PluginImplBase * *,class mavsdk::PluginImplBase * const)" (??$__std_find_trivial@PEAVPluginImplBase@mavsdk@@PEAV12@@@YAPEAPEAVPluginImplBase@mavsdk@@PEAPEAV01@0QEAV01@@Z) [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]
    D:\MAVSDK\examples\takeoff_and_land\build\Release\takeoff_and_land.exe : fatal error LNK1120: 1 unresolved externals [D:\MAVSDK\examples\takeoff_and_land\build\takeoff_and_land.vcxproj]

JonasVautherin, 2022-09-21:
Much better! So the MAVSDK you installed is a Release version, meaning that you should build your example in Release mode.

Now I suspect that you are trying to build an example from the `main` branch, but you installed MAVSDK 1.4.x. That would usually work, but we are in the process of moving to MAVSDK v2, meaning that `main` is not compatible with `1.x.x`.

Can you try to build the `takeoff_and_land` example from https://github.com/mavlink/MAVSDK/tree/v1.4.6 (instead of the one from `main`?

krr27, 2022-09-21:
MAVSDK 1.4.6 is the one I have been using.

I think I stumbled on the cause and then accidently fixed it.

Bottom line is: It now builds and runs!

I think it was an issue with the Visual Studio 2022 version I was using. 

I found the following issue report indicating that others had the same problem about a month ago, and isolated it to a VS 2022 update: (https://github.com/actions/runner-images/issues/6091)

So, I updated my Visual Studio 2022, tried again, and it worked.

Thanks again for your help!!!

Now I can move on to the real challenge - learning to use it.

closed 2022-09-21 by JonasVautherin