mavlink/MAVSDK #1903: How do you access the barometric altitude when GPS is lost?

paulholmstedt, 2022-09-21:
How do you access the barometric altitude when GPS is lost? Using the "failure gps off" command on the Linux SITL Gazebo the callback setup in with mavsdk::Telemetry::subscribe_position stop reporting values. So meaning no altitude available although I thought it was the EKF altitude reported in sbscribe_position? When GPS is lost I need to access the barometric altitude relative take-off position to aid the UAV operator when flying in joystick mode (altitude control mode). Looking at the PX4 firmware I see:
mavlink_altitude_t msg = {};
...
msg.altitude_monotonic = air_data.baro_alt_meter;
Is altitude_monotonic available in MAVSDK?
Any help much appreciated :-)

julianoes, 2022-09-22:
Ok, so we would need support for [https://mavlink.io/en/messages/common.html#ALTITUDE](ALTITUDE) I suppose.

It's funny. My goal was to come up with a subset of messages and data to make it easier and less overwhelming to figure out what one needs. But of course as time goes on, eventually all information seems to be required by someone :smile:...

So I guess this is a feature request to add altitude to telemetry.

paulholmstedt, 2022-09-23:
That would be a great feature, thanks!

closed 2022-11-02 by julianoes