mavlink/MAVSDK #1907: RC Channel Values

YasinBedirhanSimsek, 2022-10-02:
Is it possible to get any RC channel values with MAVSDK?

It doesn't matter if it is raw or scaled. Only the values of each individual RC channel is needed.

julianoes, 2022-10-02:
That's not implemented. My opinion is that higher level commands or other more type safe data should be used instead of "raw" RC channels which is usually a data specifically between the RC controller and the autopilot.

What's your use case? I'd like to know more, maybe there is another way to enable what you need.

YasinBedirhanSimsek, 2022-10-02:
I have a payload mechanism on the drone and I want to be able to control it with the RC controller (Herelink telemetry). It is connected to the Pixhawk Cube via SBUS. 

When a switch is activated, I want to capture the channel value of that switch and set the state of the mechanism (on / off).

I might be able to do this with actuator stuff in MAVSDK but the project might become more complex in the future. So more information about the RC would be better.

julianoes, 2022-10-04:
Yer, I would probably try to do it using a command, e.g. the [GRIPPER_ACTION](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_GRIPPER), however, that's a bit more involved as you rightfully pointed out.

I will put RC channels on my feature list. It's not the first time this has been requested.

If you want to make this happen before I get to it, you'd have to contribute it. Check the [docs around how to do that](https://mavsdk.mavlink.io/main/en/cpp/contributing/plugins.html#add-api-to-proto).

closed 2022-10-11 by YasinBedirhanSimsek

grybouilli, 2024-05-22:
Hello, any news on this?

YasinBedirhanSimsek, 2024-05-23:
I did it by setting a unused parameter with RC in mid-flight. I poll that parameter with mavsdk

grybouilli, 2024-05-23:
> I did it by setting a unused parameter with RC in mid-flight. I poll that parameter with mavsdk @grybouilli

Would you mind giving out a minimal reproduceable example please?

YasinBedirhanSimsek, 2024-05-23:
[image]

Take a look at this image, for example, this one is a parameter that I don't use at all.

In QGC, you can find this "Set RC to Param" button which would allow you to set any parameter with RC buttons during flight.

In mavsdk, check out the param module which will enable you to see the current parameters of the system

Simply create a blocking loop or an async timer etc... to poll the state of your selected "component" for any changes. 

I did this to control a servo that is not connected to the PX4 actuator outputs but connected to my companion computer to open and close something on the drone with RC buttons. 

If you need the exact value of an RC channel to do some sort of control etc this won't work. 

If that's your case check this out https://mavsdk.mavlink.io/main/en/cpp/api_reference/structmavsdk_1_1_telemetry_1_1_actuator_output_status.html

@grybouilli

grybouilli, 2024-05-24:
Well I implemented a PPM signal generator on a Jetson board and I want to log the values received by the Pix32 to analyze the quality of the signal. So the drone is not actually flying when I do that.
I know I can check those values in the logs that I can trigger by hand in QGC but I want to be able to either trigger the logs from a program or retrieve the values in real time.

I'll check all this and get back here when I have, thanks a lot 👍 

@YasinBedirhanSimsek