mavlink/MAVSDK #1909: 'mavsdk_server' is not distributed platform : Running MAVSDK on Raspberry Pi4 with Ubuntu 20.04

Moseskota, 2022-10-07:
Hello everyone,
I am quite new to this field. I have been trying to run MAVSDK on RPi4 with PX4_FMU_V6x connected via USB. 
As soon as gave the " await drone.connect ( ) (with or wihtout the serial port address) the prompt is throwing a comment like in the image attached. Help is appreciated.   

[image]

JonasVautherin, 2022-10-07:
And did you try following those instructions? :sweat_smile:

Moseskota, 2022-10-07:
@JonasVautherin Actually i tried to do the first step. But there was an error mentioning the mismatch between amd64 (mavsdk for ubuntu 20.04) and arm64. Any suggestions?

JonasVautherin, 2022-10-07:
And did you try getting the [`arm64`](https://github.com/mavlink/MAVSDK/releases/download/v1.4.6/mavsdk_server_linux-arm64-musl) one?

Jai-GAY, 2022-12-23:
I face an issue back in [April 2022](https://github.com/mavlink/MAVSDK-Python/issues/453#issuecomment-1108217843), after that I was tasked to work on an urgent project and did not follow up. not sure if is it related.

closed 2024-05-27 by julianoes