mavlink/MAVSDK #1910: How to query PX4 version using MAVSDK

CY-1992, 2022-10-07:
I have used MAVSDK Info plugin to query PX4 version that I am using. Info plugin gives PX4 version 1.13.0, But the exact version of PX4 is 1.13.0beta. Is there any another way to query the exact version of PX4?

julianoes, 2022-10-09:
Is the information you need not in this struct somewhere?
https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/info/include/plugins/info/info.h#L136-L148

CY-1992, 2022-10-10:
@julianoes Yes, It has some information. But If git tag of PX4 firmware is **v1.13.0-beta***, Info plugin gives version data until **v1.13.0** without **beta** as flight_sw_patch version in  [Version struct](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/info/include/plugins/info/info.h#L139) is an Int. I could use git hash instead of tag.  Some how Qgroundcontrol managed to query the PX4 version including beta using [Additional version type](https://github.com/mavlink/qgroundcontrol/blob/aa666d468f0dedc45a89d68250212e253ee668cb/src/AutoPilotPlugins/PX4/AirframeComponentSummary.qml#L37).

[image]

julianoes, 2022-10-11:
I guess the information should be somewhere and we need to decode it correctly in mavsdk.

https://github.com/PX4/PX4-Autopilot/blob/ce609144b0b8058cf4d05dac691c36c4f0a4ffd0/src/lib/version/version.c#L48-L59

https://github.com/mavlink/MAVSDK/blob/7ebcc681eaf425c5426d12bc53b529bb90062f5c/src/mavsdk/plugins/info/info_impl.cpp#L106-L108

Maybe this information is enough for you to figure it out and make a pull request? I'm pretty busy at the moment and can't look into it in detail I'm afraid.

closed 2022-10-24 by julianoes