mavlink/MAVSDK #1912: Bad include in mavlink_passthrough.h

eastman-bechamo, 2022-10-10:
Git tag: `v1.4.7`
Installation method: `cmake v3.24.2`
Platform: Ubuntu 18.04

Install steps:
`git clone https://github.com/mavlink/MAVSDK.git`
`cd MAVSDK`
`git checkout v1.4.7`
`git submodule update --init --recursive`
`/snap/bin/cmake -DCMAKE_BUILD_TYPE=Debug -Bbuild/default -H.`
`/snap/bin/cmake --build build/default -j8`
`sudo /snap/bin/cmake --build build/default --target install`

Usage in CMakeLists.txt:
`find_package(MAVSDK 1.4.7 REQUIRED)`
`target_link_libraries(... PRIVATE MAVSDK::mavsdk)`

Usage in my_header.h:
```
#include <mavsdk/mavsdk.h>
#include <mavsdk/plugins/mavlink_passthrough/mavlink_passthrough.h>
```

Compilation error:
```
In file included from /usr/local/include/mavsdk/plugins/mavlink_passthrough/mavlink_passthrough.h:8:0,
                 from /home/me/project/my_header.h:17,
/usr/local/include/mavsdk/plugins/mavlink_passthrough/mavlink_include.h:3:10: fatal error: mavlink/v2.0/common/mavlink.h: No such file or directory
 #include "mavlink/v2.0/common/mavlink.h"
```

First few lines of mavlink_passthrough.h:
```
#pragma once

#include <memory>
#include <string>
#include <functional>

// This plugin provides/includes the mavlink 2.0 header files.
#include "mavlink_include.h"
#include "mavsdk/plugin_base.h"
```

Unless I'm including the wrong file in my header, it seems like `#include "mavlink_include.h"` should be `#include "mavsdk/mavlink_include.h"` instead so that the linker (?) can find the header based on standard system include paths. I made this change locally and it resolved my issue, but this should be handled on the source-code end.

Otherwise, am I including `MavlinkPassthrough` incorrectly?

Thanks!

julianoes, 2022-10-10:
We tried to fix that in #1911 but I guess we got it wrong. Thanks for the report. I'll have a look later.

julianoes, 2022-10-10:
I'll fix this in https://github.com/mavlink/MAVSDK/pull/1913. In the meantime, I'd say just use v1.4.6.

PR #1913 (2022-10-11): [Backport v1.4] Revert #1911
I think I was wrong with #1911.
Closes #1912.
closed 2022-10-28 by eastman-bechamo