mavlink/MAVSDK #1916: MAVSDK on Jetson Nano

YasinBedirhanSimsek, 2022-10-14:
Hi everyone, 

I am currently trying to install mavsdk on Jetson Nano, i had some problems so this is how I fixed them.

The system is:

- Jetpack 4.6
- Ubuntu 18.04 LTS
- Python 3.6.9

**Because of other libraries (vpi, jetson-utils etc.) it is not possible to update python, so i am stuck with v3.6.9.** 

My first error was with pip, it didn't install the mavsdk because of a dependency

[image]

I kinda got around this problem with `pip3 install protobuf==3.19.6 mavsdk` this command, and now it can import mavsdk in python

Now the problem is the mavsdk-server, 

I connected the Pixhawk Cube - Jetson Nano via UART, and connections are ok but there is no mavsdk-server on the Jetson.

When I run the python examples:

[image]

How can i install mavsdk-server on Jetson Nano ?

Any help would be appreciated.

JonasVautherin, 2022-10-14:
> How can i install mavsdk-server on Jetson Nano ?

The Jetson Nano is arm64, isn't it? In that case you can try to just run [`mavsdk_server_linux-arm64-musl`](https://github.com/mavlink/MAVSDK/releases/download/v1.4.7/mavsdk_server_linux-arm64-musl) from the MAVSDK release page.

YasinBedirhanSimsek, 2022-10-14:
I am sorry for asking this dumb question but as i stated I am not familiar with ubuntu. I searched about it but couldn't find something about how to run such a file.

Could you please explain further or refer me to a source about how to do that?

YasinBedirhanSimsek, 2022-10-15:
I managed to run mavsdk @JonasVautherin, thanks to your response.

But it seems the mavsdk version is 0.12.0, how can I update to the latest version, pip seems to do nothing.

I already have lots of code but 0.12.0 version is not compatible.

JonasVautherin, 2022-10-15:
What makes you say it's not compatible?

YasinBedirhanSimsek, 2022-10-15:
I heavily use offboard mode and PositionGlobalYaw was not in that version it gave import errors but I managed to do some more pip stuff and now it is updated to 1.4.1. Finally I can use it.

Thank you very much for your help.

closed 2022-10-15 by YasinBedirhanSimsek

JonasVautherin, 2022-10-15:
Oh, did you do `pip install -U mavsdk` to upgrade it?

YasinBedirhanSimsek, 2022-10-16:
It didn't work at first. Turns out it was because of some other modules but I did some cleaning and `pip install -U mavsdk` did work so yes.