mavlink/MAVSDK #1917: seg fault when trying to read results of mavsdk::Param::get_all_params

weagle08, 2022-10-18:
MAVSDK version v1.4.4
PX4 v12.3

When I execute a get_all_params against the mavsdk::Param plugin I am getting a seg fault on a different thread immediately after trying to access any of the vectors of the mavsdk::Param::AllParams returned value. I'm doing the request after successfully connecting to a PX4 flight controller. I'm new to MAVSDK and mavlink in general so any assistance in tracking this down would be appreciated.

julianoes, 2022-10-18:
It would be great if you can get a stacktrace for this.

If you run this with gdb and after the crash type `backtrace`, does it show anything?

weagle08, 2022-10-19:
Working more on this.. the failure happens in a unit test (gtest).. attempted to reproduce on my end with a minimal code reproduction and the method works fine, so I’ll keep digging and close or update here when I have more info.

weagle08, 2022-10-19:
@julianoes Ok, I'm able to reproduce and have created a minimum project for you to be able to see the bug. It seems to only happen if the mavsdk::Telemetry plugin has been registered as well. It is a race condition among threads it seems, and it does not happen every time, but fairly frequently. The stack trace did not mean a lot to me, but I've included it in the repo as well as below. The README hopefully has all you need to allow you to reproduce if needed, and the build utilizes local build mavsdk instead of shared lib so you can step into the MAVSDK source if needed. It's also set up with VS Code debug if you utilize that IDE and prefer to step through that way. Ping me if you need anything further, or if you can guide me to the culprit in MAVSDK source and I can correct it.

weagle08, 2022-10-19:
If i remove the timeout guard in mavlink_parameters.cpp line 821 the issue does not appear, but I don't think I want to remove this guard and have a potentially orphaned thread.

weagle08, 2022-10-19:
I noticed there were issues in this area on main so I tested against main as well and the problem still exists.

weagle08, 2022-10-19:
I think it may potentially be this line:
https://github.com/mavlink/MAVSDK/blob/7ebcc681eaf425c5426d12bc53b529bb90062f5c/src/mavsdk/core/mavlink_parameters.cpp#L988

weagle08, 2022-10-19:
I think I've found the culprit:

https://github.com/mavlink/MAVSDK/blob/7ebcc681eaf425c5426d12bc53b529bb90062f5c/src/mavsdk/core/mavlink_parameters.cpp#L994

instead of passing back a copy of the _all_params list, the actual list is returned and so when I access it on a different thread I think this is where the seg fault happens. If this is a correct observation I can gladly put out a pull request to resolve this issue.

julianoes, 2022-10-24:
Amazing, thank you for the example! I'll try to tackle it this week.

I think the line you mentioned is not great indeed but I don't think that's the reason because in https://github.com/mavlink/MAVSDK/blob/v1.4/src/mavsdk/plugins/param/param_impl.cpp#L68 it creates a copy.

The error looks like a double fulfillment of a promise on `receive_timeout`. I thought I had fixed that one before :thinking:.

These:
https://github.com/mavlink/MAVSDK/pull/1788/commits/812fa0a94c9871b1e032cdb3c35cd1bb8ead08bf
https://github.com/mavlink/MAVSDK/pull/1788/commits/5ce3a5c2c821e5273c09a56ecf719eb5b36d52df

Now here comes the funny part. I just tried your example and couldn't reproduce it. I assume it depends on the PX4 version as well. I tested against PX4 `main`. I'll get back to this later.

weagle08, 2022-10-24:
Yes, you're right. It's definitely caused by double fulfillment of the timeout callback. That is odd that you could not reproduce the issue. I've only tested against PX4 running in a docker container. I'll try against actual hardware (CUBE H7 Blue as soon as I can and post the result).

julianoes, 2022-10-24:
I also tried against PX4 SITL v1.12.3 and it works just fine.

My best guess is that you link against some MAVSDK .so file on your system that you're not aware of. It might be worthwhile to remove anything installed system-wide in `/usr` and `/usr/local`:

```
sudo apt remove libmavsdk-dev
sudo rm -rf /usr/local/include/mavsdk /usr/local/lib/libmavsdk.so /usr/local/lib/libmavsdk.so.* /usr/local/lib/cmake/MAVSDK
```

weagle08, 2022-10-25:
I do not utilize a locally installed version of MAVSDK. I build in the same manner as the project I posted, so it does a full build of mavsdk and links against the static lib that gets output. I'll keep looking at this as I'm very curious how the callback is even being hit twice because I see the lock guards and the checks for callback being null. Will report anything I find, but may be a couple weeks.

weagle08, 2023-01-27:
I'm going to close this since I seem to be the only one that's had this issue. I haven't had a chance to address it since using a passthrough subscription for params is fine for what I'm doing for now. and if it's a non-issue I don't want it sitting open. If someone else sees the problem they'll open a new issue I'm sure.

closed 2023-01-27 by weagle08