mavlink/MAVSDK #1919: Make a vcpkg package for MAVSDK

eastman-bechamo, 2022-10-25:
Currently, I use vcpkg to manage all of the dependencies for my program. MAVSDK is the only dependency I have that can't be managed with vcpkg, and as such requires extra effort to integrate and extra effort for our customers.

Are there any plans to make a vcpkg package for MAVSDK? If there's not, I'd like to see if there's any interest, and also request that one be made.

Thank you!

JonasVautherin, 2022-10-25:
There was some interest [here](https://github.com/mavlink/MAVSDK/issues/1108). I am personally not against it, as long as it is not too intrusive in MAVSDK's build system. I don't know exactly how it works, but I would guess that it should be possible to add vcpkg without altering the CMakeLists (too much), right?

ShevaDavid, 2025-08-19:
That would simplify Windows installation significantly