mavlink/MAVSDK #1924: Provided MAVLink headers are installed to the wrong directory

eastman-bechamo, 2022-11-04:
When supplying MAVLink C-header files and building from source, the build system installs the headers to a location that's not able to be found by `mavlink_include.h`.

Ubuntu 18.04 LTS
CMake v3.24.3

Configuration:
`cmake -B build/default -S . -DMAVLINK_DIALECT=all -DMAVLINK_HEADERS=../my_headers -H.`

Build and Install:
`sudo cmake --build build/default --target install -j8`

Resulting directory structure:
`/usr/local/include/mavsdk/plugins/mavlink_passthrough/**my_headers**/mavlink/v2.0/...`

Include line from `mavlink_passthrough/mavlink_include.h`:
`#include "mavlink/v2.0/all/mavlink.h`

---

To fix this, I edited `mavsdk/src/mavsdk/plugins/mavlink_passthrough/CMakeLists.txt`:
```
install(DIRECTORY
    ${MAVLINK_HEADERS} --> ${MAVLINK_HEADERS}/mavlink
    DESTINATION ${CMAKE_INSTALL_INCLUDEDIR}/mavsdk/plugins/mavlink_passthrough
)
```

---

Please let me know if I'm doing something incorrectly. If I'm not, please update that `install(...)` command in the next release. Thanks! :)

julianoes, 2022-11-06:
Which version is that?

eastman-bechamo, 2022-11-07:
1.4.6

julianoes, 2022-11-07:
Hehe, look at the changelog for 1.4.7: https://github.com/mavlink/MAVSDK/releases/tag/v1.4.7 :smile:

closed 2022-11-07 by julianoes

eastman-bechamo, 2022-11-08:
Ahh sorry! Thanks for pointing that out :smile:

julianoes, 2022-11-14:
Although I'm confused. I think I was supposed to publish v1.4.8. I'll do that now.

julianoes, 2022-11-18:
Ok, so I broke it a second time :astonished:

julianoes, 2022-11-18:
Did 1.4.7 work?

eastman-bechamo, 2022-11-18:
Unfortunately, it didn't. So to summarize:
- v1.4.6 copies that parent directory of the custom MAVLink headers instead of the `mavlink/` folder
- v1.4.7 copies the correct `mavlink/` folder, but to a location that's inaccessible to `mavlink_passthrough`

---

Configuration:
`cmake -B build/default -S . -DMAVLINK_DIALECT=all -DMAVLINK_HEADERS=../my_headers -H.`

Build and Install:
`sudo cmake --build build/default --target install -j8`

---

#### v1.4.6

Resulting directory structure:
`/usr/local/include/mavsdk/plugins/mavlink_passthrough/**my_headers**/mavlink/v2.0/...`

Include line from mavlink_passthrough/mavlink_include.h:
`#include "mavlink/v2.0/all/mavlink.h`

---

#### v1.4.7

Resulting directory structure:
`/usr/local/include/mavsdk/**my_headers**/...`

Include line from mavlink_passthrough/mavlink_include.h:
`#include "mavlink/v2.0/all/mavlink.h`

julianoes, 2022-11-20:
Thanks, so both are wrong :thinking:

julianoes, 2022-11-20:
Can you test this branch? https://github.com/mavlink/MAVSDK/pull/1931

eastman-bechamo, 2022-11-21:
@julianoes Looks like that branch worked, thanks for fixing that :)

julianoes, 2022-11-22:
Thanks for testing, I will merge that then and make a new point release.