mavlink/MAVSDK #1932: Reading the Status of RC switches with MavlinkPassthrough

Decerguyer, 2022-11-24:
Hi, I currently have a setup with a drone, RC switch, and companion computer.
I want to be able to read the status of RC Channel 8 on the companion computer, so I have the companion computer listen in on mavlink commands with mavlinkpassthrough.

I used the subscribe_async command to receive RC_CHANNEL related RC messages. My code looks as follows:
[image]

For some reason, the output has no output like this:
[image]

On the bright side, the companion computer is receiving mavlink messages, the value is just always zero regardless of the switch position.
It should be noted I also tried reading the voltage_battery using the system_status mavlink message, and that is reading zero as well despite the real voltage being around 22.7V.

I don't know why every real reading I try to receive is zero, and would appreciate some guidance with this.
Thanks

julianoes, 2022-11-24:
Can you share a minimal example for me to reproduce this?

Alternatively, check the message in QGroundControl -> Analyze tools -> MAVLink inspector and check what the value is there.

Decerguyer, 2022-11-24:
Try to give this code a shot. Thanks for the help

void minimal_example(){
    std::cout << "Minimal Example" << std::endl;
    Mavsdk mavsdk;
    //ConnectionResult connection_result = mavsdk.add_any_connection(argv[1]);
    ConnectionResult connection_result = mavsdk.add_any_connection("serial:///dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AQ02BVZE-if00-port0:57600");
    if (connection_result != ConnectionResult::Success)
    {
        std::cerr << "Connection failed: " << connection_result << '\n';
    }
    auto system = get_system(mavsdk);
    auto action = new mavsdk::Action(system);
    auto offboard = new mavsdk::Offboard(system);
    auto telemetry = new mavsdk::Telemetry(system);
    auto mavlink_passthrough = new mavsdk::MavlinkPassthrough(system);

    mavlink_passthrough->subscribe_message_async(65, [](const mavlink_message_t &msg_raw)
{
    const mavlink_message_t* msg = &msg_raw;
    std::cout << "Recieved RC Transmission: " << std::endl;

    mavlink_rc_channels_t *rc_channels;
    mavlink_msg_rc_channels_decode(msg, rc_channels);

    std::cout << "Number of channels received: " << int(rc_channels->chancount) << std::endl;
    std::cout << "Channel 8: " << rc_channels->chan8_raw << std::endl;
}  
);
while(1);
}

std::shared_ptr<mavsdk::System> get_system(mavsdk::Mavsdk &mavsdk)
{
    std::cout << "Waiting to discover system...\n";
    auto prom = std::promise<std::shared_ptr<System>>{};
    auto fut = prom.get_future();

    // We wait for new systems to be discovered, once we find one that has an
    // autopilot, we decide to use it.
    mavsdk.subscribe_on_new_system([&mavsdk, &prom]()
                                   {
        auto system = mavsdk.systems().back();

        if (system->has_autopilot()) {
            std::cout << "Discovered autopilot\n";

            // Unsubscribe again as we only want to find one system.
            mavsdk.subscribe_on_new_system(nullptr);
            prom.set_value(system);
        } });

    // We usually receive heartbeats at 1Hz, therefore we should find a
    // system after around 3 seconds max, surely.
    if (fut.wait_for(seconds(3)) == std::future_status::timeout)
    {
        std::cerr << "No autopilot found.\n";
        return {};
    }

    return fut.get();
}

julianoes, 2022-12-04:
Ok, so channel 8 is 0. Are the other channels 1..7 different from 0?

closed 2024-05-27 by julianoes