mavlink/MAVSDK #1936: Bug in MissionRawImpl::set_current_mission_item_async()

steveb2014, 2022-11-29:
Found this check in MissionRawImpl::set_current_mission_item_async method.  the && should be an || and should be an easy fix.
----
if (index < 0 && index >= _mission_progress.last.total) {
        _parent->call_user_callback([callback]() {
            if (callback) {
                callback(MissionRaw::Result::InvalidArgument);
                return;
            }
        });
    }
------

JonasVautherin, 2022-11-29:
Do you feel like opening a PR for it, so that you get the credit? :innocent:

steveb2014, 2022-11-29:
Wow.  That's a great idea.  Never done that on a public project before.  
Will have to investigate!

JonasVautherin, 2022-11-30:
Just ping me here if you need help or decide to give up. But I think it's worth it. It's something like this:

1. Fork the repo
2. Clone your forked repo (that will be your `origin` remote)
3. Create a new branch for your fix, say `git branch fix-mission-set-current-item`
4. Commit your changes, and push to github. In the terminal, you will see instructions for opening a PR (basically a link to open in your browser). If you miss that and open your repo in github, the UI will offer you to create a PR.
5. Create a PR that points to mavlink/MAVSDK. Bonus: in the PR description, write "Resolves #1936".
6. That's it, we should see it here!

julianoes, 2022-12-03:
Whoops, good catch @steveb2014.

closed 2023-02-02 by JonasVautherin