mavlink/MAVSDK #1941: takeoff_and_landing example build error

luketjones99, 2022-12-09:
Hi, I saw that there was a previous issue and discussion going over an error building the takeoff_and_landing example. I went through that discussion a few times and I'm still a little confused. (Apologies I'm completely new to PX4 and MAVSDK development). I cloned the most recent version of the repo and got everything setup properly, but I had the same issues when trying to run the takeoff and land example.

I was able to get it to build and work in simulation when I switched to the older v1.4.3 branch. But I was just wondering if there was something I should do so that I can run the examples with the newest code?

It was mentioned that there is a CI system that should keep everything up to date, and I'm wondering if that is something I need to run manually to update the example files? (Apologies - I'm not familiar with how a CI system works). Just wanted to make sure I had everything up to date before I start practicing with my own examples. 

To put it more concisely: Is there a way to update the example file so that it will build with the newest main code? Or did I misunderstand the discussion regarding the previous issue? 

Thanks!

JonasVautherin, 2022-12-10:
If you install MAVSDK from a package, you should use the examples that correspond to its version. Say you install MAVSDK 1.4.7, you should run the examples from tag `v1.4.7` (and not from `main`).

If you compile MAVSDK from source, you should run the corresponding examples. Say you compile MAVSDK from `main`, you should use the examples on `main`.

If you try to use the examples in `main` with an older version (e.g. v0.13), then it may fail because MAVSDK has evolved since then.

Does that make sense?

luketjones99, 2022-12-10:
Thanks so much for the reply! I compiled from source (am using the latest version) and used the examples on main, but those examples seemed to still have the outdated api calls? I may still be missing something...my apologies!

JonasVautherin, 2022-12-10:
That should not happen, I think, because the CI does that :thinking:. But without more details on exactly what you did and what compilation errors you have, it's difficult to help more

luketjones99, 2022-12-10:
I was following the instructions on this quickstart guide: https://mavsdk.mavlink.io/main/en/cpp/quickstart.html. I just cloned the latest version of the repository and get all the PX4 sim stuff setup separately then I cd into the examples directory and run:

cd takeoff_and_land/
cmake -Bbuild -H.
cmake --build build -j4

The first cmake call runs properly, but the second one results in the following errors:

 error: no type named 'NewSystemHandle' in 'mavsdk::Mavsdk'
    Mavsdk::NewSystemHandle handle = mavsdk.subscribe_on_new_system([&mavsdk, &prom, &handle]() {
    ~~~~~~~~^
 error: no member named 'unsubscribe_on_new_system' in 'mavsdk::Mavsdk'; did you mean 'subscribe_on_new_system'?
            mavsdk.unsubscribe_on_new_system(handle);
                   ^~~~~~~~~~~~~~~~~~~~~~~~~
                   subscribe_on_new_system
/usr/local/include/mavsdk/mavsdk.h:301:10: note: 'subscribe_on_new_system' declared here
    void subscribe_on_new_system(const NewSystemCallback& callback);
         ^

I was able to get everything running by switching to an old branch and then re running the cmake lines, and then running:

build/takeoff_and_land udp://:14540

But, I figured if I was having to switch to an old branch to get something working, then I was doing something wrong.

JonasVautherin, 2022-12-12:
> But, I figured if I was having to switch to an old branch to get something working, then I was doing something wrong.

You are not. You should use the examples that correspond with the version of MAVSDK you installed, that's expected :+1:

closed 2022-12-12 by JonasVautherin

luketjones99, 2022-12-13:
The thing was that I installed the latest version (not the 1.4.3 version that I had to switch to to get it to work), that's why I wasn't sure.