mavlink/MAVSDK #1945: Get entries stalls indefinitely

eastman-bechamo, 2022-12-15:
**Version**: 1.4.9

I'm having an issue where getting entries from a Cube Orange stalls indefinitely half the time. Here's some example code:
```
auto logfiles = make_shared<LogFiles>(system);
auto entries = logfiles->get_entries();
/** stalls here indefinitely */
```
This code stalls as well:
```
auto logfiles = make_shared<LogFiles>(system);
vector<Entry> entries{};
Result res{Result::Next};
auto entries_cb = [&](Result r, vector<Entry> ve)->void{
    res = r;
    if (res == Result::Success) entries = ve;
};
logfiles->get_entries_async(entries_cb);
while (res != Result::Success && res == Result::Next) { sleep_for(10us); }
```

Am I doing something wrong? For reference, I have about 40 entries on the Cube.

eastman-bechamo, 2022-12-15:
After deleting the all of the log entries on the Cube, the code no longer stalls. The issue should be addressed though.

JonasVautherin, 2022-12-15:
Is that running PX4 or Ardupilot?

eastman-bechamo, 2022-12-16:
The Cube is loaded with PX4 v1.12.3

JonasVautherin, 2022-12-16:
Sounds like a bug, yeah :laughing:

julianoes, 2024-05-27:
I think that's fixed now.

closed 2024-05-27 by julianoes