mavlink/MAVSDK #1946: How can I use an older version of MAVLink with MAVSDK?

eastman-bechamo, 2022-12-20:
I'm interfacing with a vehicle that's using MAVLink messages that are outdated when compared to the current definitions for some messages. I've compiled the outdated MAVLink definitions into headers, and pointed MAVSDK to the location for those headers so they would be used:

`$(CMAKE) -B build/default -S . -DMAVLINK_DIALECT=all -DMAVLINK_HEADERS=../mavlink-headers`

I get an error during compilation of the `system_impl.cpp` file telling me that enums that are being checked for don't exist:

```
MAVSDK/src/mavsdk/core/system_impl.cpp: In member function ‘std::pair<mavsdk::MavlinkCommandSender::Result, mavsdk::MavlinkCommandSender::CommandLong> mavsdk::SystemImpl::make_command_ardupilot_mode(mavsdk::SystemImpl::FlightMode, uint8_t)’:
MAVSDK/src/mavsdk/core/system_impl.cpp:914:24: error: ‘MAV_TYPE_VTOL_TAILSITTER_DUOROTOR’ is not a member of ‘MAV_TYPE’
         case MAV_TYPE::MAV_TYPE_VTOL_TAILSITTER_DUOROTOR:
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
MAVSDK/src/mavsdk/core/system_impl.cpp:915:24: error: ‘MAV_TYPE_VTOL_TAILSITTER_QUADROTOR’ is not a member of ‘MAV_TYPE’
         case MAV_TYPE::MAV_TYPE_VTOL_TAILSITTER_QUADROTOR:
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
MAVSDK/src/mavsdk/core/system_impl.cpp:917:24: error: ‘MAV_TYPE_VTOL_FIXEDROTOR’ is not a member of ‘MAV_TYPE’
         case MAV_TYPE::MAV_TYPE_VTOL_FIXEDROTOR:
                        ^~~~~~~~~~~~~~~~~~~~~~~~
MAVSDK/src/mavsdk/core/system_impl.cpp:918:24: error: ‘MAV_TYPE_VTOL_TAILSITTER’ is not a member of ‘MAV_TYPE’
         case MAV_TYPE::MAV_TYPE_VTOL_TAILSITTER:
                        ^~~~~~~~~~~~~~~~~~~~~~~~
```

Is there a way around this problem? Conditional compilation? Compile-time static checks? Is there a 'hack' that I could use to get this working quickly?

CMake: `v3.25.1`
Ubuntu: 18.04

Thanks!

JonasVautherin, 2022-12-20:
If you change the version of MAVLink, you change the generated headers. If those headers break the compilation, you have to fix it. So in this case you would have to fork MAVSDK and go fix `system_impl.cpp`, because apparently this older version of MAVLink you are using is not compatible with the MAVSDK sources you compile.

I think `-DMAVLINK_HEADERS` is more here to allow you to have an extended version of MAVLink, such that you can use your custom messages from mavlink_passthrough. But the other way (going back to an older version) is not guaranteed to work, as apparently you are showing here :slightly_smiling_face:

eastman-bechamo, 2022-12-21:
I wonder if there's interest in allowing older versions of MAVLink? I don't think it would be too hard, except for the fact that MAVLink is maybe not very stringent with its versioning.

JonasVautherin, 2022-12-21:
I don't think there is any kind of reliable versioning at all, is there? Other than v1/v2 :thinking:

eastman-bechamo, 2022-12-21:
I didn't think so either. I think it's something that MAVLink could do better. It would be useful for allowing tools to communicate with different protocols without changing source code.

julianoes, 2022-12-22:
MAVLink broke the build here but not the protocol. So if you have an older MAVLink version where you use the "Reserved" enum it will be binary compatible.

We had a big discussion on the MAVLink call whether "reserved" enums/defines should have even been used in source or whether they are "reserved" and not to be used. There are pros and cons, it's not clear cut.

You can see the history here: https://github.com/mavlink/mavlink/pull/1818

closed 2024-05-27 by julianoes