mavlink/MAVSDK #1947: "WARNING: THIS FILE IS AUTOGENERATED" comments in plugin files

twolski83, 2022-12-21:
Some files (example mavsdk\plugins\mission\include\plugins\mission\mission.h) have following comment:

"// WARNING: THIS FILE IS AUTOGENERATED! As such, it should not be edited.
// Edits need to be made to the proto files
// (see https://github.com/mavlink/MAVSDK-Proto/blob/main/protos/mission/mission.proto)" 

As I quickly check the code this comment made me confused as I understand they are not autogenerated. Is my assumption correct?

JonasVautherin, 2022-12-21:
> Is my assumption correct?

Probably not :sweat_smile:. If there is the warning saying that the file is autogenerated, it means that it is autogenerated :see_no_evil:

twolski83, 2022-12-21:
Thanks,
First time in code. Now I see where it comes from.

closed 2022-12-21 by twolski83

julianoes, 2022-12-22:
See: https://mavsdk.mavlink.io/main/en/cpp/contributing/autogen.html