mavlink/MAVSDK #1948: Directly import a mission as character data

blank-supportgis, 2022-12-22:
I'm currently developing a mission management for an autonomous drone in Python, using MAVSDK's Python bindings. A mission is sent as a string to the drone's companion computer via LTE, gets imported using the command `drone.mission.import_qgroundcontrol_mission` and is then uploaded to the flight controller.
The inconvenience here is, that the function `import_qgroundcontrol_mission` only accepts a filename as input, so I have to write the received mission string to a file, Import that file and delete it again afterwards.This is especially inconvenient since most companion computers will use SD cards as their memory, which are notoriously short lived when receiving frequent write operations. Every write operation to disk you can avoid should be avoided.

Could we have an alternative import operation that accepts the plan data directly as string? I attempted it myself, but been struggling with correctly building everything for some time now. For a developer who has the build environment readily set up and understands the workings of MAVSDK, the effort should be very small.

julianoes, 2022-12-22:
Thanks for the issue. Good idea! And I like how you provide some context of why this would be helpful for you!

julianoes, 2022-12-22:
@blank-supportgis any chance you could test this? https://github.com/mavlink/MAVSDK/pull/1950

blank-supportgis, 2022-12-23:
> @blank-supportgis any chance you could test this? #1950

Thanks, that was quick!
I don't know if I can test it this year. Like I said, I'm struggling with building everything, but that is problably a story for another thread...

PR #1950 (2022-12-24): mission_raw: add method to import plan from string
This enables to import mission plan files without having to write them to some location first.

Depends on https://github.com/mavlink/MAVSDK-Proto/pull/305.

Closes #1948.
closed 2022-12-24 by julianoes