mavlink/MAVSDK #1956: mavsdk_server build fail

dayleMG, 2023-01-12:
I want build mavsdk_server, but there had some problems.. 

I have already installed this one 
```
sudo apt-get install build-essential cmake git
```

I used this command There was a no problem
```
cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=OFF -DBUILD_MAVSDK_SERVER=ON -Bbuild/default -H.
```

But this one had problems
```
cmake --build build/default -j8
```

[image]

Looks like grpc library error.. So I have installed grpc with this guide
https://grpc.io/docs/languages/cpp/quickstart/
But still occur errors.

My Ubunt environment 

cmake version 3.16.3

OpenSSL 1.1.1i  8 Dec 2020

Distributor ID: Ubuntu
Description:    Ubuntu 20.04.3 LTS
Release:        20.04
Codename:       focal

JonasVautherin, 2023-01-12:
Is it a clean build? Looks like an issue with OpenSSL. Did OpenSSL build successfully above?

dayleMG, 2023-01-12:
Thank you for comment @JonasVautherin 

[image]
[image]
Umm I'm not sure about that this one is end point of OpeenSSL.

And there has warnings.
It looks like there has some problems.
[image]
[image]
[image]

JonasVautherin, 2023-01-12:
Can you search for "Found OpenSSL" in your log? It should find the one that you just built, not the one on your system.

Also just to be sure: are you building the `main` branch?

julianoes, 2023-01-12:
And what sort of Ubuntu are you using?

dayleMG, 2023-01-13:
Thank you for comment @julianoes 

This one is my Ubuntu environment.

Distributor ID: Ubuntu
Description: Ubuntu 20.04.3 LTS
Release: 20.04
Codename: focal

dayleMG, 2023-01-13:
@JonasVautherin 
Yes my branch is this one [image]

Do you mean this log?
[image]

JonasVautherin, 2023-01-13:
> Found OpenSSL: /usr/local/lib/libcrypto.so

This is wrong, it should find it in `/preneu/utm/MAVSDK/build/default/third_party/install/lib`. Can you check that you have the ssl libraries installed in `/preneu/utm/MAVSDK/build/default/third_party/install/lib`?

julianoes, 2023-01-13:
@dayleMG and is that latest `main`?

You can do `git rev-parse HEAD` to check the commit.

dayleMG, 2023-01-16:
@JonasVautherin 
[image]
Do you mean this one??

dayleMG, 2023-01-16:
@julianoes 
Here you are~ 444d6b693b2d76cd594b1df074f28f468cc78f47

JonasVautherin, 2023-01-16:
> Do you mean this one??

Yes. I don't see libcrypto and libssl there, it feels like OpenSSL failed to build/install :thinking:

closed 2023-01-30 by dayleMG

julianoes, 2023-01-30:
So what was the fix? Sorry I missed it.

dayleMG, 2023-01-31:
@julianoes that's okay thank you
I found mavsdkserver releases artifacts. so I have just used that one. 
Thanks for helping me.

tbago, 2023-05-31:
I use ubuntu 22.04 to build the mavsdk server. And also cannot found the openssl library. 
First the openssl library in ubuntu 22.04 is output at **build/third_party/install/lib64** dir.
And the openssl seems only generate pkg-config file and no cmake config file.

julianoes, 2023-06-01:
@tbago please create a new separate issue and describe the steps you do and what goes wrong, thank you.