mavlink/MAVSDK #1957: How to know if offboard.set_position_global completed sucessfully

mjemv, 2023-01-20:
As the return value of the API only denotes success of message sent to pixhawk, not the completion of the API.
There is no way to know if the drone has reached to the destination Lat/Long. 

Should we need to monitor telemetry position callback to check if the drone is on right path and do an estimation based on speed and angle and predict lat long ?

Please guide

-Thanks

JonasVautherin, 2023-01-20:
Are you sure that you want the offboard mode? Feels like you maybe want `action.goto_location` instead? Which doesn't confirm when the position is reached either, so you have to use the telemetry :innocent:.

But offboard is a more advanced mode of PX4.

mjemv, 2023-01-20:
Telemetry Position has to be constantly monitored in the callback to check if the drone is on track and reached destination.
Moreover there could be a slight delta in actual destination lat,long,alt.

Why isn't there a broadcast in Mavlink similar to MISSION_CURRENT for this case.
How long would the GCS wait before giving next goto_location, assuming that we are giving a defined mission but an on the fly fine grained lat long controlled by keyboard keys.

JonasVautherin, 2023-01-20:
I guess that's a PX4 question :sweat_smile:. I don't know how difficult it would be for PX4 to have a concept of "goto"...

mjemv, 2023-01-20:
How long would the GCS wait before giving next goto_location, that drone has reached the destination.
How do you handle this in the code that uses MAVSDK?

closed 2023-01-21 by mjemv