mavlink/MAVSDK #1959: How to arm drone without GPS

mjemv, 2023-01-21:
We are trying to do a BVLOS operation indoors, in this case we dont have a GPS lock but LTE is available.
But without GPS arming is not possible even in offboard mode.

How to get over this in MAVSDK if GPS lock is not available.

julianoes, 2023-01-22:
Hm, arming should still be possible, however, not in position control.

Offboard mode would have to be attitude control, unless you have something for a position estimate, such as flow. For BVLOS I would be convinced that you need something that helps you with position estimation indoors, otherwise this would be very hard fly.

mjemv, 2023-01-23:
Offboard test works on jmavsim but not on pixhawk hardware. 
Getting GPS bad fix error

Please suggest how to handle this
```
`
nogps_test$ ./noGpsTest tcp://192.168.1.8:14550
[01:14:25|Info ] MAVSDK version: v1.4.0-360-g66a69e9d (mavsdk_impl.cpp:24)
Waiting to discover system...
[01:14:25|Debug] New system ID: 1 Comp ID: 1 (mavsdk_impl.cpp:611)
[01:14:25|Debug] Component Autopilot (1) added. (system_impl.cpp:382)
[01:14:25|Warn ] Vehicle type changed (new type: 2, old type: 0) (system_impl.cpp:232)
[01:14:25|Debug] Discovered 1 component(s) (system_impl.cpp:551)
Discovered autopilot
Starting Offboard attitude control
Offboard started
Waiting for system to be ready
Waiting for system to be ready
Waiting for system to be ready
Waiting for system to be ready
Waiting for system to be ready
Waiting for system to be ready
Waiting for system to be ready
**[01:14:42|Debug] MAVLink: critical: PreArm: GPS 1: Bad fix (system_impl.cpp:258)**
Waiting for system to be ready

`
```

Firmware Mavproxy logs

```
AP: ArduCopter V4.2.2 (4fcfa4b2)
AP: ChibiOS: 93e6e03d
AP: fmuv3 003C0020 35385113 31333431
AP: RCOut: PWM:1-14
AP: IMU0: fast sampling enabled 8.0kHz/1.0kHz
AP: Frame: QUAD/X
...
AP: PreArm: GPS 1: Bad fix
```

CODE
`...

    auto system = get_system(mavsdk);
    if (!system) {
        return 1;
    }

    // Instantiate plugins.
    auto action = Action{system};
    auto offboard = Offboard{system};
    auto telemetry = Telemetry{system};

    {
        std::cout << "Starting Offboard attitude control\n";

        // Send it once before starting offboard, otherwise it will be rejected.
        Offboard::Attitude roll{};
        roll.roll_deg = 30.0f;
        roll.thrust_value = 0.6f;
        offboard.set_attitude(roll);

        Offboard::Result offboard_result = offboard.start();
        if (offboard_result != Offboard::Result::Success) {
            std::cerr << "Offboard start failed: " << offboard_result << '\n';
            return false;
        }
        std::cout << "Offboard started\n";
[... 90 lines cut ...]
[01:19:33|Debug] MAVLink: info: Landing at current position	 (system_impl.cpp:258)
Vehicle is landing...
[01:19:33|Debug] MAVLink: info: Landing at current position	 (system_impl.cpp:258)
Vehicle is landing...
Vehicle is landing...
[01:19:35|Debug] MAVLink: info: Preflight Fail: No manual control input	 (system_impl.cpp:258)
Vehicle is landing...
Vehicle is landing...
Vehicle is landing...
Vehicle is landing...
Vehicle is landing...
Vehicle is landing...
[01:19:42|Debug] MAVLink: critical: Preflight Fail: High Accelerometer Bias (system_impl.cpp:258)
[01:19:42|Debug] MAVLink: info: Preflight Fail: No manual control input	 (system_impl.cpp:258)
Vehicle is landing...
Vehicle is landing...
[01:19:44|Debug] MAVLink: info: Preflight Fail: No manual control input	 (system_impl.cpp:258)
Vehicle is landing...
[01:19:44|Debug] MAVLink: info: Landing detected	 (system_impl.cpp:258)
Landed!
[01:19:46|Debug] MAVLink: info: Disarmed by landing	 (system_impl.cpp:258)
[01:19:46|Debug] MAVLink: info: Preflight Fail: No manual control input	 (system_impl.cpp:258)
Finished...

```

mjemv, 2023-01-23:
The modes supported by the pixhawk firmware are
```
`GUIDED> mode
GUIDED> Available modes:  dict_keys(['STABILIZE', 'ACRO', 'ALT_HOLD', 'AUTO', 'GUIDED', 'LOITER', 'RTL', 'CIRCLE', 'POSITION', 'LAND', 'OF_LOITER', 'DRIFT', 'SPORT', 'FLIP', 'AUTOTUNE', 'POSHOLD', 'BRAKE', 'THROW', 'AVOID_ADSB', 'GUIDED_NOGPS', 'SMART_RTL', 'FLOWHOLD', 'FOLLOW', 'ZIGZAG', 'SYSTEMID', 'AUTOROTATE', 'AUTO_RTL'])

GUIDED> mode GUIDED_NOGPSAP: PreArm: GPS 1: Bad fix

GUIDED> Got COMMAND_ACK: DO_SET_MODE: ACCEPTED
GUIDED_NOGPS> Mode GUIDED_NOGPS
Got COMMAND_ACK: REQUEST_AUTOPILOT_CAPABILITIES: ACCEPTED
Got COMMAND_ACK: SET_MESSAGE_INTERVAL: ACCEPTED
Got COMMAND_ACK: DO_SET_MODE: ACCEPTED
GUIDED> Mode GUIDED
Got COMMAND_ACK: REQUEST_MESSAGE: ACCEPTED
Got COMMAND_ACK: DO_SET_MODE: ACCEPTED
Got COMMAND_ACK: COMPONENT_ARM_DISARM: FAILED
`
```

I had manually set GUIDED_NOGPS, but the offboad start sets it to GUIDED.
dronekit supports a GUIDED_NOGPS though...

julianoes, 2023-01-24:
Just note that you can't compare jMAVSim with what you have on your Pixhawk as it looks like you're using ArduPilot which is a different software from PX4.

MAVSDK mostly supports PX4 but compatibility with ArduPilot is work in progress, so you'll have to work through details and debug what doesn't work.

mjemv, 2023-01-24:
I have radiolink pixhawk fmuv3. While dronekit python has a GUIDED_NOGPS, which is found also in the supported modes by firmware, there is no specific non_gps mode in px4 i believe. Is offboard mode + attitude means non_gps?

Since we are trying to make a very small and realtime code that manages mavlink, we cant use python.
That was reason for choosing C++.

We have seen some issues with MAVSDK and this pixhawk, In GPS mode, sometimes the copter arms for a second and disarms for no reason 

There are now 2 ways a) use dronekit python or b) add stubs in MAVSDK.
What would it take to add compatability with fmuv3?

ykhedar, 2023-06-02:
As @julianoes mentioned in his first comment, you are missing some sort of position estimation which could be using optical flow in indoors scenario. Without that and a proper configuration of the firmware (Ardupilot or PX4) the vehicle has no way of knowing where it is and therefore cannot go where it should. Therefore your issue seems unrelated to MAVSDK.
Closing for now.

closed 2023-06-02 by ykhedar

MalleswariReddy1593, 2023-09-05:
Hi,
i am facing this two issue Got COMMAND_ACK: VIDEO_START_CAPTURE: FAILED and
 Got COMMAND_ACK: VIDEO_STOP_CAPTURE: FAILED could you please give me solution

julianoes, 2023-09-06:
@MalleswariReddy1593 please don't necrobump and create a new issue with proper information and context.