mavlink/MAVSDK #1962: Uploading empty mission on mission_raw_server  lead to core dump

weirdgyn, 2023-01-25:
Uploading an empty mission to a mission_raw_server instance will trigger a core dump.
This's probably a faulty behaviour of the GCS itself but QGroundControl leave users sending empty missions without complains.
Unluckly in mission_raw_server_impl.cpp this function:
```
void MissionRawServerImpl::set_current_seq(std::size_t seq)
{
    if (_current_mission.size() < static_cast<size_t>(seq)) {
        return;
    }

    _current_seq = seq;

    // If mission is over, just set item to last one again
    auto item = seq == _current_mission.size() ? _current_mission.back() :
                                                 _current_mission.at(_current_seq);
    auto converted_item = convert_item(item);
    _parent->call_user_callback([this, converted_item]() {
        if (_current_item_changed_callback) {
            _current_item_changed_callback(converted_item);
        }
    });

    mavlink_message_t mission_current;
    mavlink_msg_mission_current_pack(
        _parent->get_own_system_id(),
        _parent->get_own_component_id(),
        &mission_current,
        static_cast<uint16_t>(_current_seq));
    _parent->send_message(mission_current);
}
```
such case is not trapped by the if guard on top and then:
```
    auto item = seq == _current_mission.size() ? _current_mission.back() :
                                                 _current_mission.at(_current_seq);
```
Will dump a core...

JonasVautherin, 2023-01-25:
Would you be willing to contribute a fix for that? Maybe it could return a `RESULT_INVALID_ARGUMENT` for an empty mission? Or some other error?

weirdgyn, 2023-01-25:
Maybe I can arrange a fix and submit a pull-request... but I have not idea
bout how to return  RESULT_INVALID_ARGUMENT  .