mavlink/MAVSDK #1965: subscribe_on_new_system callback never triggered

weirdgyn, 2023-01-26:
Mavsdk mavsdk;

	Mavsdk::Configuration configuration(
		Mavsdk::Configuration::UsageType::Autopilot);

	int opt;
	uint8_t _id;

	char url[256] = DEF_URL;

	while ((opt = getopt(argc, argv, "i:u:")) != -1)
	{
	   switch (opt)
	   {
		   case 'i':
			   cout << "System id: " << optarg << endl;
			   _id = (uint8_t)atoi(optarg);
			   configuration.set_system_id(_id);
			   break;

		   case 'u':
			   cout << "Connecting on: " << optarg << endl;
			   strcpy(url , optarg);
			   break;

		   default:
			   fprintf(stderr, "Usage: %s [-i system_id] [-u connection_url]\n", argv[0]);
			   exit(EXIT_FAILURE);
	   }
   }

	mavsdk.set_configuration(configuration);

	cout << "Making connection...";

	ConnectionResult conn_result = mavsdk.add_any_connection(url);

	if (conn_result != mavsdk::ConnectionResult::Success) {
	    cout << "connection failed: " << conn_result << endl;
	    return -1;
	}
	else {
		cout << "connection created" << endl;

        mavsdk.subscribe_on_new_system(
            [&](){
        		cout << "System connected"  << endl;
            }
        );
	}

      do {
         this_thread::sleep_for(std::chrono::milliseconds(1000));
      } while(1);

The callback on `subscribe_on_new_system` is never triggered (tested with QGroundControl) ...

JonasVautherin, 2023-01-26:
Note: this was already asked in https://discuss.px4.io/t/callback-on-mavsdk-subscribe-on-new-system-is-never-called/30564/2

weirdgyn, 2023-01-26:
yes I started it there also...

julianoes, 2023-01-27:
Can you give some more steps on how I should run this, and with what arguments, so that I can reproduce it?

julianoes, 2023-01-27:
This is what you have to debug:

https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/core/mavsdk_impl.cpp#L296-L302

Presumably it triggers for you, even though it shouldn't.

julianoes, 2023-01-27:
You can also debug that using `export MAVSDK_MESSAGE_DEBUGGING=1`.

julianoes, 2024-05-27:
No follow up, closing.

closed 2024-05-27 by julianoes