mavlink/MAVSDK #1967: MAVSDK Only Connects to PX4 Autopilot AFTER QGroundControl is Launched!

OliverHeilmann, 2023-01-28:
# Problem Description
Hi, I am having an issue with connecting to the autopilot on an OrangeCube flight controller (running PX4 latest stable firmware version) with MAVSDK. Strangely, MAVSDK **can** find the user defined serial port of the OrangeCube, but fails to find an autopilot when subscribing to the system.

Even more strangely, if I first open QGroundControl (QGC) and allow it to connect to the flight controller, then close QGC, then re-run the MAVSDK executable (exactly as before), the autopilot is found and a connection is successfully made! 

I have also determined that connecting the OrangeCube to QGC through my laptop, then disconnecting the OrangeCube (making sure that it still has power) and then connecting the usb cable to another machine (specifically, Raspberry Pi 4B+) will actually allow the autopilot subscription to succeed when running any MAVSDK example. My deduction here is that QGC is doing something to the OrangeCube/ PX4 firmware which allows the autopilot to be found. 

I have also found a [PX4 forum post](https://discuss.px4.io/t/connection-only-works-after-opening-qgc/19085/2) from others with the same issue. They found a work around by running a pymavlink script prior to their MAVSDK code (which I tried but didn't work). I suspect that MAVSDK does not include a connection step it the codebase (which QGC does) and could be a potential development activity for the next release.

# The Step-by-Step
After physically connecting the OrangeCube microUSB port to my laptop (```uname -a``` gives ```Linux <machine> 5.15.0-58-generic #64~20.04.1-Ubuntu SMP Fri Jan 6 16:42:31 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux``` via usb, I run the following command:

```sh
[IN] cd /dev/serial/by-id && ls
[OUT] usb-CubePilot_CubeOrange_0-if00
```

Next I go to the [Battery example code](https://github.com/mavlink/MAVSDK/tree/main/examples/battery) in MAVSDK, build it and pass in the serial usb connection, with the _baud rate = 57600_ as an input as follows:

```sh
[IN] cmake -S . -B build
[IN] make -j4 -S build/
[IN] ./build/battery serial:///dev/serial/by-id/usb-CubePilot_CubeOrange_0-if00:57600

[OUT]
[12:31:21|Info ] MAVSDK version: v1.4.7 (mavsdk_impl.cpp:20)
Waiting to discover system...
No autopilot found.
```

Next, I launch QGroundControl. Wait for the flight controller to be detected and then quit it.
```sh
[IN] ./QGroundControl
[IN] ctrl+c
```
Following the same steps as before, I re-run the battery example. The autopilot connects!
```sh
[IN] ./build/battery serial:///dev/serial/by-id/usb-CubePilot_CubeOrange_0-if00:57600

[OUT]
[12:35:02|Info ] MAVSDK version: v1.4.7 (mavsdk_impl.cpp:20)
Waiting to discover system...
[12:35:02|Debug] New: System ID: 1 Comp ID: 1 (mavsdk_impl.cpp:482)
[12:35:02|Debug] Component Autopilot (1) added. (system_impl.cpp:377)
[12:35:02|Warn ] Vehicle type changed (new type: 10, old type: 0) (system_impl.cpp:225)
[12:35:02|Debug] Discovered 1 component(s) (system_impl.cpp:578)
Discovered autopilot
disarmed
disarmed
disarmed
```

# Code Flow Analysis - MAVSDK Battery C++ Example
As the failed autopilot connection is printed to the console ```no autopilot found```, we know that the code passes the ```mavsdk.add_any_connection(argv[1])``` connection step in the code below.

```cpp
    Mavsdk mavsdk;
    ConnectionResult connection_result = mavsdk.add_any_connection(argv[1]);

    if (connection_result != ConnectionResult::Success) {
        std::cerr << "Connection failed: " << connection_result << '\n';
        return 1;
    }
```
Next, the get_system function is called...
```cpp
    auto system = get_system(mavsdk);
    if (!system) {
        return 1;
    }
```
This is where the issue is! We can see that the mavsdk.subscribe_on_new_system(...) step does not discover an autopilot in the 3 seconds which is allocated before a timeout. Feel free to bump that number up to 10 seconds, this still doesn't make a difference for me.
```cpp
std::shared_ptr<System> get_system(Mavsdk& mavsdk)
{
    std::cout << "Waiting to discover system...\n";
    auto prom = std::promise<std::shared_ptr<System>>{};
    auto fut = prom.get_future();

    // We wait for new systems to be discovered, once we find one that has an
    // autopilot, we decide to use it.
    mavsdk.subscribe_on_new_system([&mavsdk, &prom]() {
        auto system = mavsdk.systems().back();

        if (system->has_autopilot()) {
            std::cout << "Discovered autopilot\n";

            // Unsubscribe again as we only want to find one system.
            mavsdk.subscribe_on_new_system(nullptr);
            prom.set_value(system);
        }
    });

    // We usually receive heartbeats at 1Hz, therefore we should find a
    // system after around 3 seconds max, surely.
    if (fut.wait_for(seconds(3)) == std::future_status::timeout) {
        std::cerr << "No autopilot found.\n";
        return {};
    }

    // Get discovered system now.
    return fut.get();
}
```
As mentioned above, after QGC connects to the OrangeCube, you are free to connect it to any other device and run the battery example - it will work as you'd expect. This implies that QGC is changing a setting on the OrangeCube to enable autopilot subscription.

# No Issue with Earlier PX4 Firmware
I had another OrangeCube flight controller to hand running an earlier version of PX4. For completeness, I tested the same processes out with it and had no issue connecting to the autopilot with MAVSDK! Unfortunately, I upgraded the firmware and, therefore, cannot remember what version it was that worked. Hope this point supports my observations - that the issue lies somewhere in the PX4 firmware settings.

# Hardware and Software Versions
1) Linux <machine> 5.15.0-58-generic ~20.04.1-Ubuntu SMP Fri Jan 6 16:42:31 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
2) MAVSDK v1.4
3) PX4 firmware latest:stable
4) QGroundControl [v4.2.4](https://github.com/mavlink/qgroundcontrol/releases/tag/v4.2.4)

julianoes, 2023-01-28:
Thanks for the detailed issue! This is, unfortunately, a known issue that I haven't gotten around to fix yet.

The problem is that PX4 starting with v1.13 only starts sending MAVLink once it has received a heartbeat from a GCS or MAVSDK.

Unfortunately, I closed some of the issues mentioning it when I should have left them open. At least I'll now keep this one open until solved.

Also see
https://github.com/mavlink/MAVSDK/issues/1883
https://github.com/mavlink/MAVSDK/issues/1908

This workaround should work as long as you use the C++ library (and not mavsdk_server):
https://github.com/mavlink/MAVSDK/issues/1908#issuecomment-1269316153

OliverHeilmann, 2023-01-28:
Thanks for the speedy reply and for the linked issues! 

Very happy to see that there is a simple work around. For others following this, the proposed changes in #1883 and #1908 worked for me too.

Great to see the #1968 pull request as I am sure it will save others the headache I went through these past days...

julianoes, 2023-01-29:
Yer, sorry about that. Last time I tried to make the change it seemed too complex, because heartbeats are not sent in a connection but in the `MavsdkImpl`, so I didn't see any easy way without a refactor. Today, I realized there was an easier way.

closed 2023-01-29 by julianoes

julianoes, 2023-01-29:
@OliverHeilmann could you possibly test if the code I threw out here actually works, so either `main` or `v1.4` branch? If so, I'll then do a release.

OliverHeilmann, 2023-01-30:
@julianoes I just built and tested the v1.4 branch while connected to the OrangeCube (running latest PX4 version) and it works! 

For anyone wishing to follow these steps I was testing #1969 and I followed the build instructions linked on the [MAVSDK documentation](https://mavsdk.mavlink.io/main/en/cpp/guide/build.html). I am running Linux so my steps were as follows:

Download the source using git:
```sh
git clone https://github.com/mavlink/MAVSDK.git
```
Make sure to get all the submodules as well:
```sh
git submodule update --init --recursive
```
Configure first, then build:
```sh
cmake -DCMAKE_BUILD_TYPE=Debug -Bbuild/default -H.
cmake --build build/default -j8
```
To install system-wide the command needs to be run with sudo on Linux/macOS:
```sh
sudo cmake --build build/default --target install
```
> :warning: **Conflicting builds**: If you have already installed MAVSDK before!
I had a mavsdk install in the ```/usr/include``` directory, not the ```/usr``` directory as mentioned in the documentation. Similarly, the build library ended up in the ```/usr/local/include``` directory. I deleted the install found in the ```/usr/include``` section to avoid conflicts.

julianoes, 2023-01-30:
Nice! Thanks you so much @OliverHeilmann. I will cut a release then.

OliverHeilmann, 2023-01-31:
Just tested [1.4.11 release](https://github.com/mavlink/MAVSDK/releases/tag/v1.4.11) on a Raspberry Pi and the updates work as expected.

PR #1968 (2023-01-29): core: always send heartbeat on serial connections
PX4 starting with v1.13 does not send heartbeats by default, so we need to initiate the MAVLink connection by sending heartbeats. Therefore, we override the default here and enable sending heartbeats.

Fixes #1967.