mavlink/MAVSDK #1974: Using MAVSDK with only Iridium connection

lvanasse, 2023-01-31:
Hi! I'm building a drone that would only have a SATCOM connection to it. And I want to be able to send commands to it using SATCOM. And while testing with MAVSDK I notice that it seems to require a connection to PX4 before being able to send MAVLink commands, is that correct? 

So I was wondering if it is possible to put the MAVSDK_server into a mode that would only give back the MAVLink string (in hex) so I can then send it through the SATCOM website and communicate with my drone. I would also like reverse path also, so decoding a MAVLink string and return a human-readable string.

Thanks in advanced!

JonasVautherin, 2023-01-31:
> it seems to require a connection to PX4 before being able to send MAVLink commands, is that correct?

* When you start the connection with `udp://:14540`, it's equivalent to `udp://0.0.0.0:14540`, and it says "have MAVSDK _listen_ (/bind to) on 14540". In that mode, the drone must send MAVLink messages to MAVSDK on port 14540.
* When you start the connection with `udp://192.168.1.12:14555`, then it tells MAVSDK to start sending MAVLink heartbeats to the drone (here `192.168.1.12`) on port 14555. In that mode, the drone must listen (bind) to 14555.

Note that I invented the IPs and ports, you can use whatever you want. The point is just that `udp://:<port>` and `udp://0.0.0.0:<port>` run mavsdk in "server" mode, and `udp://<some_ip>:<port>` run mavsdk in "client" mode.

Does that help?

lvanasse, 2023-01-31:
Hi @JonasVautherin! I'm familiar with the concept of using the UDP connection and port in general. But I'm more asking where the drone is only able to communicate with it through SATCOM, something like Iridium. So there's no real interface possible, either through UDP or Serial. Imagine that the drone is at 60 000ft (or 25 km) of altitude :sweat_smile: and we want to send and receive MAVLink messages. There's already the Iridium driver in PX4, but I need to add the part that decode and encode those messages.  

In an ideal world, I'd like that the MAVSDK_server should just generate it's MAVLink message for the command I'm asking it for and would decode a MAVLink message that I've provided to it. We want to put that MAVSDK_server in our backend. Let me know if that's more clear :)!

JonasVautherin, 2023-01-31:
Oh... sorry I read to quickly :sweat_smile:. I don't know if it would make sense to add support for Iridium in MAVSDK, but could you make a bridge, say like this?

```
[------ on the ground -----]                                                [--- on the drone ---]
MAVSDK <--- udp ---> bridge     <------------ Iridium --------------->       drone
```

So you write your UDP-to-Iridium proxy on the ground, and connect MAVSDK to it over UDP. For MAVSDK, it's just a UDP connection, and the drone just receives MAVLink messages.

Then the question would be whether or not MAVSDK generates too much traffic for your connection, I guess?

lvanasse, 2023-02-01:
No worries, and I wasn't able to add a label to the issue, but it was more of a question towards MAVSDK if it was possible or not. So if it doesn't fit into the design that's alright, I'm more looking at a solution for what I want to do :).

As per your suggestion, that might be a good idea. In the Iridium driver in the PX4 documentation, they actually recommand something similar to what you're suggesting (http://docs.px4.io/main/en/advanced_features/satcom_roadblock.html#overview), here's the diagram:
[image]

But we really want to control what's going out of the MAVSDK_server, like on the PX4 we are only allowing some MAVLink streams  going out of the SATCOM since sending through SATCOM is costly : /.

I'd really like to use MAVSDK for this, but I'm unsure if it's a good fit, since I want something more like an encoder/decoder for MAVLink. Maybe I can just add a mode when starting the MAVSDK_server where it's kind of offline and behave differently?

julianoes, 2023-02-01:
Basically MAVSDK expects some heartbeats from the drone. Are you saying you would not have heartbeats? If so, we would have to make a config option that allows to have the timeout much bigger. 

However, you sorta need to discover a system initially, otherwise you don't know what the target system ID and target component ID to send a command to.

MAVSDK also does a bit of initial back and forth to get some basic information. We would have to have some "low bandwidth" mode to avoid all that but it would require some work.

JonasVautherin, 2023-02-01:
It feels quite specific, I'm not sure if it would fit in MAVSDK. It sounds like you really need to control what goes out of MAVSDK, so anyway I would personally do that with a proxy. Still it may be worth using MAVSDK instead of re-implementing the protocols (e.g. mission transfer, commands, etc).

You can already increase the timeout for commands in MAVSDK. Then for the heartbeat, one way would be to have your proxy "pretend to be the drone" and send regular heartbeats to MAVSDK (say your proxy sends a heartbeat every second, and receives the heartbeats from the drone at whatever rate you want). Same for others, like the parameters: you could maybe have a cache in your proxy, in order to save round-trips.

> MAVSDK also does a bit of initial back and forth to get some basic information.

I also would like to know the minimal set of messages that MAVSDK needs to get started. It needs the heartbeats, and then at least a few params (I think). I may get to make a list in the next couple of weeks, so that could help. If you do before me, feel free to share it here :blush:

lvanasse, 2023-02-01:
Hi @julianoes! 
> Basically MAVSDK expects some heartbeats from the drone. Are you saying you would not have heartbeats? If so, we would have to make a config option that allows to have the timeout much bigger.

We would mainly use a modified version of the [HIGH_LATENCY2](https://mavlink.io/en/messages/common.html#HIGH_LATENCY2) for most of our communication with the GCS. So they would act as heartbeat for us.

> However, you sorta need to discover a system initially, otherwise you don't know what the target system ID and target component ID to send a command to.
MAVSDK also does a bit of initial back and forth to get some basic information. We would have to have some "low bandwidth" mode to avoid all that but it would require some work.

Yeah, that was my understanding while using it with the PX4 SITL. And that was a motivation behind the questions I'm asking. I feel like MAVSDK is more supported for PX4 than the alternatives, but I wanted to know if it does what I want it to do :sweat_smile:.

closed 2023-02-02 by lvanasse