mavlink/MAVSDK #1977: Blanket blacklist/whitelist MAVlink msgids

devbharat, 2023-02-03:
Do you think it makes sense to add support for white/blacklisting incoming and outgoing mavlink message by their IDs?

I am think about a new user who wants to adopt mavsdk as a library (say, for some specific list of non flight-critical features) but doesn't necessarily 'trust' it to not effect the flight behavior of their system, then instead of going into the implementation and commenting out code or enabling/disabling, they could use this 'coverall' filter to narrow down what can come in or go out of the library.

JonasVautherin, 2023-02-04:
I think I would do that out of MAVSDK, in a proxy (something like a mavlink-router that would have a whitelist). Doesn't mavlink-router allow that already?

devbharat, 2023-02-06:
makes sense

closed 2023-02-06 by devbharat

julianoes, 2023-02-07:
I would say you can already do this using the `intercept_` methods in `Mavsdk`:

https://github.com/mavlink/MAVSDK/blob/7708d66371a77c297f86d89fc7c07330fac01a10/src/mavsdk/core/include/mavsdk/mavsdk.h#L355-L381

You just return `true` in the callback to keep the message, or `false` to block it.

---

And a comment on the side: I would call it blocklist/passlist to use more inclusive language.

devbharat, 2023-02-08:
nice!