mavlink/MAVSDK #1978: Tag `v1.4.12` removed?

ZhongRuoyu, 2023-02-04:
Hi! Homebrew contributor here. We noticed that tag `v1.4.12` was removed. Was that intended?

julianoes, 2023-02-07:
Sorry, I pushed it by mistake. It was a duplicate of v1.4.11.

chenrui333, 2023-02-08:
@julianoes thanks for confirming that was intended, any timeline plan for a new release any time soon (asking so that we don't have to downgrade the mavsdk, which is some painful process for homebrew side). Thanks a lot!

julianoes, 2023-02-08:
Ok, I'll try to make a release tomorrow to make this easier for you. Thanks!

chenrui333, 2023-02-09:
That would be much better.

julianoes, 2023-02-10:
I've released v1.4.12, hopefully that works for you.
https://github.com/mavlink/MAVSDK/releases/tag/v1.4.12

closed 2023-02-10 by julianoes

ZhongRuoyu, 2023-02-10:
Thanks @julianoes (and @chenrui333)!

julianoes, 2023-02-10:
And thank you for maintaining the homebrew updates!