mavlink/MAVSDK #1981: Install interface in cmakelist

devbharat, 2023-02-07:
I am curious why some plugins(camera_server, camera_information...) specify 

```
$<INSTALL_INTERFACE:include/mavsdk>
```

while most others specify

```
$<INSTALL_INTERFACE:include>
```

in their CMakeLists.txt. I am asking because I see some weird behavior similar to [this](https://github.com/mavlink/MAVSDK/issues/1912) if I disable all plugins of the former category (`INSTALL_INTERFACE:include/mavsdk`) via plugins.txt.

JonasVautherin, 2023-02-07:
Hmm it sounds like those plugins may be wrong, I don't see why they would be different :thinking:

julianoes, 2023-02-07:
Does this work? https://github.com/mavlink/MAVSDK/pull/1982

I'm not quite sure which variant is correct, sadly.

devbharat, 2023-02-07:
I have a feeling it needs to go the other way (keep the include/mavsdk in the plugins), or include the mavsdk headers in the plugin headers with the full path (mavsdk/mavsdk_xyz.h) but not sure, will check tomorrow!

devbharat, 2023-02-08:
>I have a feeling it needs to go the other way (keep the include/mavsdk in the plugins), or include the mavsdk headers in the plugin headers with the full path (mavsdk/mavsdk_xyz.h) but not sure, will check tomorrow!

@julianoes yes that works. That's how i would have tried it as well, perhaps I didn't see the diff right earlier. Thanks for the fix.

closed 2023-02-08 by devbharat

julianoes, 2023-02-08:
I changed it and force pushed. Sorry, that would have been confusing!