mavlink/MAVSDK #1988: Trouble with Visual Studio 2019 and the "takeoff_and_land" example.

aspidron-cd, 2023-02-10:
Hello!

I'm not sure if this is an IDE specific problem or not. I am new to this. Please forgive me.

I'm using the prebuild mavsdk-windows-x64-release.zip from [1.4.6] on Windows 10 .0.1945.
I extracted the zip to a local folder "C:\SharedLibs\mavsdk_superbuild".
I tried the takeoff_and_land example from the 1.4.6 tree.
I'm using the VS 2019 Version 16.10.4 IDE and set the project properties (Release/x64) as follows:

C/C++: Additional Include Directories: C:\SharedLibs\mavsdk_superbuild\include
C/C++: C++ Language Standard: ISO C++17 Standard (/std:c++17)
Linker: Additional Library Directories: C:\SharedLibs\mavsdk_superbuild\lib
Linker: Additional Dependencies: mavsdk.lib

Building the example in the IDE (Release/x64) throws me:

>Build started...
1>------ Build started: Project: takeoff_and_land, Configuration: Release x64 ------
1>mavsdk.lib(system_impl.obj) : error LNK2001: unresolved external symbol __std_find_trivial_8
1>C:\Users\Chris\source\repos\takeoff_and_land\x64\Release\takeoff_and_land.exe : fatal error LNK1120: 1 unresolved externals
1>Done building project "takeoff_and_land.vcxproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========
<

Obviously the linker isn't happy. What am I doing wrong?

aspidron-cd, 2023-02-10:
Update: Using VS 22 instead works perfectly. 

In case only VS22 is supported now the installation guide on dronecode.org is outdated and misleading.

JonasVautherin, 2023-02-10:
> In case only VS22 is supported now the installation guide on dronecode.org is outdated and misleading.

Would you mind updating the installation guide? I guess it's fine to point people to VS22.

julianoes, 2023-02-10:
Sorry about that. Yes please create a PR to update the instructions.

CI already uses 2022:
https://github.com/mavlink/MAVSDK/blob/d3115f547f8a7dee4b6c61dea1211ca80101d4b9/.github/workflows/main.yml#L610

aspidron-cd, 2023-02-11:
Actually Visual Studio 2019 is only mentioned in the instructions for building the SDK from source, which most Windows users might want to skip on first contact. I could add instructions on how to build the examples in VS with precompiled libraries if that's helpful to anyone. But this might bloat the guide with IDE specific stuff.
Anyway, this issue can be closed I guess.

closed 2023-02-11 by JonasVautherin

julianoes, 2023-02-12:
Alright, I'll do the PR then: https://github.com/mavlink/MAVSDK-docs/pull/270.