mavlink/MAVSDK #1989: MAVSDK verson on macOS aarch64

cryptik, 2023-02-17:
I am working on a Mac M1 (aarch64) and was able to install mavsdk from homebrew, which installs mavsdk_server version 1.4.12.  However, this does not install the Python3 bindings.  When trying to install mavsdk-python from PyPi via `pip3 install mavsdk`, the only versions available are 0.5.2, 0.7.0, 0.8.0, 0.8.1, 0.8.2, 0.9.0, 0.9.1, 0.10.0, 0.11.0, 0.12.0.  None of these versions include the mavsdk_server executable.

I am sure it is possible to compile the latest version (1.4.12) of the python bindings, but I am not sure how to do that or how to install it into my python virtual environment (miniconda) once it is built.

julianoes, 2023-02-18:
Yes, that's currently broken. The main pain point is that GitHub actions doesn't have macOS M1/2 runners, and so it's not easy to release the binary.

Besides that, something is wrong with how the architecture gets downloaded or is compatible and so the recent releases don't work.

Also see:
https://github.com/mavlink/MAVSDK-Python/blob/29f2502d03c540a21270651610ebbbe72cfcc8f5/setup.py#L55-L73

https://github.com/mavlink/MAVSDK-Python/issues/319

cryptik, 2023-02-25:
@julianoes quick question... so I was able to install the aarch64 version of mavsdk_server v1.4.12 via home-brew... but this only includes the C++ interface library, not the python version. Can I build/install the mavsdk_python package from the source?  I am not sure how this is done as I have never built a python package that could then be installed via pip. Do you know where I can find a guide on how to build the pip package?  

Also, looking at the two posts you provided above... here... 

```
        elif sys.platform.startswith('darwin'):
            return 'macos'
```
This only checks if the platform is macOS.. and not the architecture. This probably defaults to x86.  The code for Linux checks the architecture.  I can probably fix that and submit a pull request if I can figure out how to build/install the python package.  Any pointers would be most helpful.

cryptik, 2023-02-25:
@julianoes scratch my last.  I was able to clone the repo and followed the instructions at the bottom of the README.md.  When I first saw them, I did not realize that they build the python sdk.  I was expecting something talking about building a wheel and what not.  I have never built a python package from source.

Anyway, the instructions worked flawlessly and I was able to build MAVSDK version 1.4.12 for aarch64 on M1.  Seems to work just fine. :)

JonasVautherin, 2023-02-26:
That's great! Are you detecting the M1 around [here](https://github.com/mavlink/MAVSDK-Python/blob/main/setup.py#L67-L68)?

Then the problem we have is that we don't have a CI for the M1 that could build `mavsdk_server` and release it. I wish that the github CI had that, but it doesn't at the moment...

cryptik, 2023-02-27:
So for my current immediate need, I used `mavsdk_server` installed from `homebrew,` and then used the `MAVSDK` repo instructions to build the python bindings (as you know, it won't create the executable for MacOS aarch64).  Both versions numbers match at this point.

The code block that you highlighted is one of the problems.  It appears to detect that the platform is MacOS (i.e., Darwin) but does not check the architecture.  I am also still determining how a python library is built (i.e., setup.py) for distribution, so I need to understand how that works.  I use them a lot.. just never had to build one :)

JonasVautherin, 2023-02-27:
> So for my current immediate need, I used mavsdk_server installed from homebrew

Oh, so homebrew actually builds MAVSDK for the M1 and ships the binary? Is there a way to download that binary directly (say with `curl`)? Because then we could leverage that in the CI: MAVSDK-Python could ship with the homebrew binary for M1...

Maybe @ZhongRuoyu or @chenrui333 would have an insight about that? It would solve our problem. Right now we fetch `mavsdk_server` from our Github releases (that's done [here in setup.py](https://github.com/mavlink/MAVSDK-Python/blob/main/setup.py#L100)).

cryptik, 2023-02-27:
Perfect! The homebrew install path differs for M1 (`/opt/homebrew/bin` vs. `/usr/local/bin/brew`), but these paths are added to the user's PATH when they install brew.  So, for the M1 install script, it would be easy to search for `mavsdk_server` in the `/opt/homebrew/bin` folder. This is a great idea!

I use the `mavsdk-python` bindings inside a Conda environment, so having the server executable outside of the site_packages folder structure is a better setup for me... however, users will need to be mindful that the `pip` installed package may have a different version number than the brew installed executable.  I don't think this is a big deal... but it could cause strange behavior if the versions differ significantly.

ZhongRuoyu, 2023-02-28:
> Oh, so homebrew actually builds MAVSDK for the M1 and ships the binary?

Yes, we do. The pre-built binaries are called [bottles](https://docs.brew.sh/Formula-Cookbook#homebrew-terminology), and they are hosted [here](https://github.com/Homebrew/homebrew-core/pkgs/container/core%2Fmavsdk) on GitHub.

> Is there a way to download that binary directly (say with curl)? Because then we could leverage that in the CI: MAVSDK-Python could ship with the homebrew binary for M1...

You may use `brew fetch mavsdk` to download that. To force a specific bottle, use e.g. `brew fetch --bottle-tag=arm64_monterey`. The bottle tags are the ones shown in the [formula](https://github.com/Homebrew/homebrew-core/blob/master/Formula/mavsdk.rb).

But please note that bottles are not necessarily ready for use; they need to be *poured* (processed) by Homebrew before that. [^1] Here's what you'll likely see when download the fetched bottle, extract it, and run the executable.

```
$ ./bin/mavsdk_server
dyld[73411]: Library not loaded: @@HOMEBREW_PREFIX@@/opt/openssl@1.1/lib/libssl.1.1.dylib
  Referenced from: <8EF3AA53-B6AC-3F6F-B75F-7F0C22B73FC6> /Users/ruoyu.zhong/Downloads/mavsdk/1.4.12_1/lib/libmavsdk_server.1.4.12.dylib
  Reason: tried: '/System/Volumes/Preboot/Cryptexes/OS@@HOMEBREW_PREFIX@@/opt/openssl@1.1/lib/libssl.1.1.dylib' (no such file), '/usr/local/lib/libssl.1.1.dylib' (no such file), '/usr/lib/libssl.1.1.dylib' (no such file, not in dyld cache)Library not loaded: @@HOMEBREW_PREFIX@@/opt/tinyxml2/lib/libtinyxml2.9.dylib
  Referenced from: <5F93883E-9924-325F-8523-A9DCA0DA4D6C> /Users/ruoyu.zhong/Downloads/mavsdk/1.4.12_1/lib/libmavsdk.1.4.12.dylib
  Reason: tried: '/System/Volumes/Preboot/Cryptexes/OS@@HOMEBREW_PREFIX@@/opt/tinyxml2/lib/libtinyxml2.9.dylib' (no such file), '/usr/local/lib/libtinyxml2.9.dylib' (no such file), '/usr/lib/libtinyxml2.9.dylib' (no such file, not in dyld cache)
[1]    73411 abort      ./bin/mavsdk_server
```

[^1]: Although this is not the case for `mavsdk`, some bottles may not be relocatable. That means they will only work when installed to the same location as when they were built. (But `mavsdk`'s macOS bottles are relocatable, as they are tagged [`:any`](https://docs.brew.sh/Bottles#cellar-cellar)).

JonasVautherin, 2023-02-28:
@ZhongRuoyu: thanks a lot for the insights! So I guess the best way is to tell the user to install it with homebrew when the M1 is detected (instead of automatically trying to fetch the executable).

julianoes, 2024-06-02:
Done in https://github.com/mavlink/MAVSDK/pull/2277.

closed 2024-06-02 by julianoes