mavlink/MAVSDK #1992: Can't get telemetry with subscibe_position()

eugeneknvlv, 2023-02-22:
Hi,

I'm actually 2 days in drones, trying to get started :) My friend lent me a pixhawk and gps module "Cirocomm T0010". Now i'm trying to get coordinates - latitude and longitude, but getting nothing.

I took the takeoff_and_land code from examples and just commented the part I don't need, got smth like this:

```
...
using namespace mavsdk;
using std::chrono::seconds;
using std::this_thread::sleep_for;

void usage(const std::string& bin_name)
{
    std::cerr << "Usage : " << bin_name << " <connection_url>\n"
              << "Connection URL format should be :\n"
              << " For TCP : tcp://[server_host][:server_port]\n"
              << " For UDP : udp://[bind_host][:bind_port]\n"
              << " For Serial : serial:///path/to/serial/dev[:baudrate]\n"
              << "For example, to connect to the simulator use URL: udp://:14540\n";
}

std::shared_ptr<System> get_system(Mavsdk& mavsdk)
{
    std::cout << "Waiting to discover system...\n";
    auto prom = std::promise<std::shared_ptr<System>>{};
    auto fut = prom.get_future();

    // We wait for new systems to be discovered, once we find one that has an
    // autopilot, we decide to use it.
    mavsdk.subscribe_on_new_system([&mavsdk, &prom]() {
        auto system = mavsdk.systems().back();

        if (system->has_autopilot()) {
            std::cout << "Discovered autopilot\n";

            // Unsubscribe again as we only want to find one system.
            mavsdk.subscribe_on_new_system(nullptr);
            prom.set_value(system);
        }
    });

    // We usually receive heartbeats at 1Hz, therefore we should find a
    // system after around 3 seconds max, surely.
    if (fut.wait_for(seconds(3)) == std::future_status::timeout) {
        std::cerr << "No autopilot found.\n";
        return {};
    }

    // Get discovered system now.
    return fut.get();
}

int main(int argc, char** argv)
{
    if (argc != 2) {
        usage(argv[0]);
        return 1;
    }

    Mavsdk mavsdk;
    ConnectionResult connection_result = mavsdk.add_any_connection(argv[1]);

    if (connection_result != ConnectionResult::Success) {
        std::cerr << "Connection failed: " << connection_result << '\n';
        return 1;
    }

    auto system = get_system(mavsdk);
    if (!system) {
        return 1;
    }

    // Instantiate plugins.
    auto telemetry = Telemetry{system};
    //auto action = Action{system};

    // We want to listen to the altitude of the drone at 1 Hz.
    const auto set_rate_result = telemetry.set_rate_position(1.0);
    if (set_rate_result != Telemetry::Result::Success) {
        std::cerr << "Setting rate failed: " << set_rate_result << '\n';
        return 1;
    }

    // Set up callback to monitor altitude while the vehicle is in flight
    telemetry.subscribe_position([](Telemetry::Position position) {
        std::cout << "Altitude: " << position.relative_altitude_m << " m\n";
   	std::cout << "Longitude: " << position.longitude_deg << '\n';
	std::cout << "Latitude: " << position.latitude_deg << '\n';
     });

    return 0;
}
```

I checked if gps module works using QGroundControl and it seems to be fine. In MAVLink Inspector I see the right coordinates of gps module. But subscribe_position still does nothing.

Also tried getting position using blocking methods like:
```
std::cout << telemetry.position().latitude_deg << '\n'; 
```
and
```
std::cout << telemetry.raw_gps().latitude_deg << '\n';
```
The first one prints nan, the second one prints 0. 
No idea how to deal with it.

eugeneknvlv, 2023-02-22:
The same thing with all the telemetry. Running e.g. `std::cout << telemetry.attitude_euler().roll_deg << '\n'` gives me `nan` too.

julianoes, 2023-02-22:
What PX4 or ArduPilot version are you running on the Pixhawk?

eugeneknvlv, 2023-02-22:
Finally overcame it myself :)
Thanks a lot you for your responsiveness though

closed 2023-02-22 by eugeneknvlv

julianoes, 2023-02-23:
What was the problem? For someone else finding the same...