mavlink/MAVSDK #1994: How to do calibration of  quadcopter using MAVSDK

mjemv, 2023-02-27:
Hi,

QGC/MP have a process of calibrating ESC/Magnetometer/Accelerator etc 
For an application we have to provide this functionality so we cant use QGC etc.
MAVSDK offers a limited functionality and there is a single api with limited progress data.
Mavlink offers MAV_CMD_PREFLIGHT_CALIBRATION and MAG_CAL_REPORT etc.

I am stuck how to do it with MAVSDK, is it implemented ? or I will have to use MAVLINK messages, in that case is there a sample?

JonasVautherin, 2023-02-27:
Did you try the [example](https://github.com/mavlink/MAVSDK/blob/main/examples/calibrate/calibrate.cpp)?

JonasVautherin, 2023-02-27:
Did you try running the example to see what happens? :see_no_evil:

The example will print steps ("now please put the pixhawk on the side") and instructions. You would then just need to connect that to a UI, I think.

julianoes, 2024-05-27:
No follow up, closing.

closed 2024-05-27 by julianoes