mavlink/MAVSDK #1996: Ardupilot and MAVSDK compatability

41Mo, 2023-03-02:
# Bug report

Greetings. I am trying to communicate with ArduCopter firmware via MAVSDK, and i found out a few problems.

**Issue details**

1. action.takeoff() implementation fopr apm - [takeoff_async_apm](https://github.com/mavlink/MAVSDK/blob/ab24200b9df9176890ec5e80c332708ba72d2a59/src/mavsdk/plugins/action/action_impl.cpp#L387)  function works only if vehicle is not in offboard mode. This is kinda strange behaviour. e.g. User might change mode to offboard via rc controller and then want to takeoff, but in this case command would not be sent.
Probably there is missing else branch.

Another strange thing, not a bug, but misleading behavior is telemetry.set_rate_battery(). This function sets MAVLINK_MSG_ID_SYS_STATUS, which is used not only for battery, but for health status primarily. Probably this function should be called set_rate_sys_status()? And for battery rate MAVSDK should send [BATTERY_STATUS](https://mavlink.io/en/messages/common.html#BATTERY_STATUS).

julianoes, 2023-03-05:
Hi @41Mo, thanks for the notes.

1. This was added by @ykhedar. Let's see if he can explain why it's done that way. I don't have insight into it.
2. That's indeed misleading. Especially now that we also parse the proper battery status message, as far as I remember, so we could fix this now properly. Would you mind creating a pull request to fix it?

ykhedar, 2023-03-05:
Hi @41Mo Good catch regarding point 1. It is indeed missing the else block. Do you want to create a PR for this? Otherwise, I can do that in a few days.

41Mo, 2023-03-06:
@julianoes I will be happy to do pr on 2 issue.

Jai-GAY, 2023-10-24:
> # Bug report
> Greetings. I am trying to communicate with ArduCopter firmware via MAVSDK, and i found out a few problems. Some of them already discussed in #1990 issue.
> 

Have you successfully [get more actions working with Arducopter](https://github.com/mavlink/MAVSDK-Python/issues/453#issuecomment-1774507019)? does the [set_rate_battery helps](http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/plugins/telemetry.html#mavsdk.telemetry.Telemetry.set_rate_battery)? if yes, what value (hz) do you use?

julianoes, 2026-03-31:
I'm sure there are more, so I'd be inclined to leave this open as a catch all.