mavlink/MAVSDK #2001: MavFrame enum is not correct?

chengguizi, 2023-03-12:
In the file https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/mocap/include/plugins/mocap/mocap.h

I see MavFrame is defined like the following, which is used to fill up the ODOMETRY message for MAVLINK. I assume they are arrange so `MocapNed` is coded as 0 and `LocalFrd` is coded as 1

```
        enum class MavFrame {
            MocapNed, /**< @brief MAVLink number: 14. Odometry local coordinate frame of data given
                         by a motion capture system, Z-down (x: north, y: east, z: down).. */
            LocalFrd, /**< @brief MAVLink number: 20. Forward, Right, Down coordinate frame. This is
                         a local frame with Z-down and arbitrary F/R alignment (i.e. not aligned
                         with NED/earth frame). Replacement for MAV_FRAME_MOCAP_NED,
                         MAV_FRAME_VISION_NED, MAV_FRAME_ESTIM_NED.. */
        };
```

However, in MAVLINK they are coded both at 20. Is this something wrong here? Shouldn't we just expose the API as the native MAVLink enum, instead of MAVSDK custom ones?

https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/mocap/include/plugins/mocap/mocap.h#L293

julianoes, 2023-03-12:
The MAVLink frames are messy and there are two many of it, so my goal was to only expose a smaller but working subset in MAVSDK.

However, I think you are right that the implementation is not correct. It seems like it's just hard-coded?

https://github.com/mavlink/MAVSDK/blob/e7231dc8f390b96f4682e6810ac3d2fe554b5b8b/src/mavsdk/plugins/mocap/mocap_impl.cpp#L223

Would you mind making a pull request to fix it?

chengguizi, 2023-03-13:
Sure @julianoes 

Actually the line you just quoted has no issues, ok to be hardcoded. The issue is the line above.

https://github.com/mavlink/MAVSDK/blob/e7231dc8f390b96f4682e6810ac3d2fe554b5b8b/src/mavsdk/plugins/mocap/mocap_impl.cpp#L222

I will do pull requests once I verify things on a real PX4 system :)