mavlink/MAVSDK #2002: Work to get v2.0 out the door

julianoes, 2023-03-15:
I have been working on various refactoring and changes towards the next major revision. For visibility, let me list the remaining work that I would like to get done before shipping v2:

- [ ] Better logging, easily customizable and redirectable
- [ ] Rename `_parent` to e.g. `_system_impl` everywhere
- [x] Rename all `MAVLinkSomething` to `MavlinkSomething` for consistency
- [ ] Split FTP client and server
- [ ] Clean up param receiver and server (WIP: https://github.com/mavlink/MAVSDK/pull/1772)
- [ ] Consolidate/improve configuration options
- [x] Add waypoint actions
- [ ] Try both: REQUEST_MESSAGE and older custom getters
- [ ] Clean up duplicate plugin constructors
- [ ] Fix/disable failing CI tests

Consti10, 2023-03-31:
Let me recommend you spdlog for logging

JonasVautherin, 2023-03-31:
Looking into them, I think that quite a few are not breaking the API, are they? I agree that those are nice improvements, but they could be done after the v2 release, as implementation details.

For instance:
* Fix/disable failing CI tests
* Try both: REQUEST_MESSAGE and older custom getters
* Rename _parent to e.g. _system_impl everywhere

Or do you think they are really required before v2?

julianoes, 2023-03-31:
@Consti10 
> Let me recommend you spdlog for logging

Or, alternatively, I was hoping we can go to C++20 soon and use `std::format`: https://en.cppreference.com/w/cpp/utility/format.

julianoes, 2023-03-31:
> Looking into them, I think that quite a few are not breaking the API, are they?

Yer, you're right. We can always tackle them later too.

tlapik123, 2023-06-05:
> @Consti10
> 
> > Let me recommend you spdlog for logging
> 
> Or, alternatively, I was hoping we can go to C++20 soon and use `std::format`: [en.cppreference.com/w/cpp/utility/format](https://en.cppreference.com/w/cpp/utility/format).

I'll just add to this that the latest gcc/clang compiler will need to be used for <format> to work. (not sure if that's an issue, but it's good to know).
[image]
[source](https://en.cppreference.com/w/cpp/compiler_support#C.2B.2B20_library_features)

julianoes, 2023-06-06:
@tlapik123 right, so maybe not yet. I usually would require the version that comes with the most recent Ubuntu LTS and latest MSVC installation. Sadly with Ubuntu 22.04 we're stuck with GCC 11.3 (or GCC 12 if installed) and Clang 14.

Consti10, 2023-06-07:
you can easly compile fmt from sources drectly when using spdlog (it does so by default).
Then gcc fmt support is not needed.

julianoes, 2023-06-07:
spdlog seems like the obvious choice then.

holden-zenith, 2023-07-29:
Maybe an issue similar to this one specifically for Ardupilot compatibility changes needed would be useful. I have been busy on other projects, but soon I hope to work more on this so I will make note of what I notice needs to be done here for now.

- [Follow Me] Config parameters for Ardupilot are different, so this plugin won't work until they are implemented. Capabilities also currently only implements POS. Need to also adjust follow mode detection.

julianoes, 2023-08-07:
> [FTP] Returns Protocol Error when trying to grab parameters

This is something we'll have to address. Have you tried with the #2060 PR? I'm reworking FTP there.

julianoes, 2023-08-07:
> [General] Don't repeatedly request unsupported messages

Sure, let's work through those and fix them. Also not blocking v2.

julianoes, 2023-08-07:
>do you have any reservations about having a port override in the shell plugin?

I'm not sure what exactly you need and why. I suggest you open a new issue and describe further what you mean.

julianoes, 2023-12-27:
Done and dusted: https://github.com/mavlink/MAVSDK/releases/tag/v2.0.0

closed 2023-12-27 by julianoes