mavlink/MAVSDK #2003: Move docker images away from dockerhub

julianoes, 2023-03-15:
It sounds like dockerhub is no longer hosting the docker images for free, if I read the email right:

> If you own a legacy Free Team organization, access to paid features — including private repositories — will be suspended on April 14, 2023 (11:59 pm UTC). Upgrade your subscription before April 14, 2023 to continue accessing your organization. 

So it sounds like we need to find a way around this. Either paying, using the AWS docker registry (paid), or maybe Github Actions provides something?

JonasVautherin, 2023-03-15:
Right, MAVSDK seems to be a Docker Free Team (same as px4io). Should we try to apply for their OSS sponsorship? https://www.docker.com/community/open-source/application/

julianoes, 2023-03-15:
I suppose we should.

JonasVautherin, 2023-03-16:
Or we just migrate to ghcr :sweat_smile:

julianoes, 2023-03-25:
Nevermind, docker reversed their decision. We can leave it as is.

closed 2023-03-25 by julianoes