mavlink/MAVSDK #2013: REQUEST_MESSAGE vs custom getters like REQUEST_CAMERA_INFORMATION

JonasVautherin, 2023-03-31:
It's currently part of the [list for MAVSDK v2](https://github.com/mavlink/MAVSDK/issues/2002), and though I don't think I can contribute an implementation, I think I can contribute a (relatively informed) opinion.

TL;DR: I don't think this can go in MAVSDK v2 :see_no_evil:.

#### We cannot detect what the receiver supports

There is no way to detect if the receiver supports `REQUEST_MESSAGE`, because the requested message could have been requested by someone else, or it could be transmitted spontaneously by the receiver. So making statistics to try to guess stuff like "was this `CAMERA_INFORMATION` message an answer to my request, or is it a coincidence?" is probably not worth it by far.

Also each message is independent. So even if we managed to detect that `REQUEST_MESSAGE` is supported for `CAMERA_INFORMATION`, it would not say anything about e.g. `REQUEST_PROTOCOL_VERSION`.

#### What we can do

* Push the receivers to support both messages (i.e. the drone should support both `REQUEST_MESSAGE(<camera_information>)` and `REQUEST_CAMERA_INFORMATION`.
* Keep the old way in MAVSDK until we feel like we can completely drop it.

In other words, the implementation work is not on the MAVSDK side, but on the receiver side (e.g. make sure that PX4/Ardupilot support the newer messages, make sure that cameras do, push proprietary players to do it).

Some day, maybe we will feel comfortable in dropping the old way and telling people "well if your drone does not support the new way, that's your problem".

julianoes, 2023-04-02:
Thanks for bringing this up. Let's talk through it and make a decision.

My question is: Wouldn't a receiver that doesn't support `REQUEST_MESSAGE` nack it and therefore tell you that it doesn't support it?

JonasVautherin, 2023-04-03:
> Wouldn't a receiver that doesn't support REQUEST_MESSAGE nack it

I don't think so :thinking:. One problem with MAVLink is that the specs do not require to  nack unsupported messages. I am pretty sure I have seen implementations that just ignore unsupported messages (resulting in a timeout, indistinguishable from an actual timeout in a sane way IMO).

So same thing there: one would have to make it a requirement to nack unsupported messages, ask implementations to change to that, and in a few years when we are fairly sure that all the implementations we care about do it, we could rely on nacks :innocent:.

julianoes, 2023-04-03:
So actually UNSUPPORTED is for a command that is not supported but not necessarily for a command param that is not supported. So, in that case, it should probably be failed. I would argue a decent camera implementation would do that, rather than timing out.

So what about switching to the new REQUEST_MESSAGE command and if it gets UNSUPPORTED, FAILED or a timeout, then we fall back to the old one?
This means that old implementations would still work, just slower, while new ones will work nicely, and quickly. What am I missing?

JonasVautherin, 2023-04-04:
Actually, maybe one thing we could do is always use both from MAVSDK. I think that could work :thinking:.

julianoes, 2023-04-10:
> But I see many problems there

Yes, there are many problems. But at some point switching and saying we don't support the old way also doesn't work and also has problems. So my idea is to try hard and do a best effort solution that - while having corner cases - generally works.

> For instance, it is legal to broadcast the command and expect multiple systems/components to broadcast their message.

Yes, it gets indeed tricky with multiple systems and components, so let's only use these commands to specific sysid/compid pairs and not use broadcast.

> Should we somehow keep a state in MAVSDK that says "I am pretty sure that this drone supports the new message because I got an answer once"?

I think you're missing something here. A success is not if we receive the message that we asked for but we receive the ack *AND* the message. That's a definite confirmation that it worked.

julianoes, 2023-04-11:
Yes, exactly!

julianoes, 2024-05-27:
In v3 I want to remove the [MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES](https://mavlink.io/en/messages/common.html#MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES).

PR #2381 (2024-09-01): Switch to new REQUEST_MESSAGE but fall back to deprecated request message commands if required
This implements the logic of requesting messages using the new command as well as the deprecated commands consolidated in the RequestMessage class.

Currently WIP implementing the functionality in REQUEST_MESSAGE. The next step is to use this class consistently everywhere.

Closes #1605, closes #2013.
closed 2024-09-01 by julianoes