mavlink/MAVSDK #2015: MAVSDK equivalent of POSITION_GLOBAL_TARGET_INT?

TheKeveloper, 2023-04-05:
This is a noob question and I'm probably just missing something in the MAVSDK docs, but is there a way to tell the current target of the drone using MAVSDK? Basically what we would get from the [POSITION_GLOBAL_TARGET_INT](https://mavlink.io/en/messages/common.html#POSITION_TARGET_GLOBAL_INT) message?

JonasVautherin, 2023-04-05:
Something like [`action.goto_location()`](https://github.com/mavlink/MAVSDK-Proto/blob/main/protos/action/action.proto#L76-L84)?

TheKeveloper, 2023-04-05:
@JonasVautherin Sorry, maybe I wasn't super clear. I know we can use that to tell the drone where to go, but there's no way for us to inspect the state of the drone and find out where it's current target is if that makes sense?

We could do something like maintain some state for where the drone is going each time we send a go to location command, but it seems like Mavlink already provides a message type that tells us the current target of the drone.

JonasVautherin, 2023-04-05:
Hmm maybe @julianoes can answer that. My limited understanding is that `goto_location` puts PX4 in "hold mode", with the setpoint set there. I don't think that there is a notion of "target reached" (or even a notion of "target" at all).

Not sure how Ardupilot does it though. What autopilot are you using?

TheKeveloper, 2023-04-05:
I'm using the px4 gazebo simulator: https://github.com/JonasVautherin/px4-gazebo-headless. I seem to be receiving this message: https://mavlink.io/en/messages/common.html#POSITION_TARGET_GLOBAL_INT, when communicating directly over mavlink.

I'm thinking of switching over to MAVSDK but couldn't find how to support this using mavsdk

JonasVautherin, 2023-04-05:
Maybe we could add this message to the `telemetry` plugin? Something like `SubscribePositionTargetGlobal`?

julianoes, 2023-04-10:
The equivalent is [Offboard::set_global_position](https://mavsdk.mavlink.io/main/en/cpp/api_reference/classmavsdk_1_1_offboard.html#classmavsdk_1_1_offboard_1a97b2bd8f516e267e5b28373f369ff8d3).

Also see:

https://github.com/mavlink/MAVSDK/blob/ad478f19ab99fce4a9b16f33353306f45764acf1/src/mavsdk/plugins/offboard/offboard_impl.cpp#L448-L470

julianoes, 2023-04-10:
Ok, and if you want to actually see tthe telemetry if someone else is commanding this, then that's not supported.

You could use MavlinkPassthrough and subscribe to https://mavlink.io/en/messages/common.html#POSITION_TARGET_GLOBAL_INT.

TheKeveloper, 2023-04-10:
@julianoes thanks for the response! Yeah, that makes sense. I'm using the mavsdk java client for my workflow currently so can't access the mavlink passthrough. Not sure if this is something that may be worth adding to the proto?

JonasVautherin, 2023-04-10:
I think it would be worth adding to the telemetry plugin. It does make sense to me that a ground station may want to show where the drone is headed in that situation :innocent:

julianoes, 2023-04-11:
Fair enough, feel free to make a pull request.

julianoes, 2023-04-11:
And make sure to make PR to the [MAVSDK-Proto](https://github.com/mavlink/MAVSDK-Proto/) repo first.

julianoes, 2026-09-15:
With v4 and MavlinkDirect you'll be able to access any message as JSON.

closed 2026-09-15 by julianoes