mavlink/MAVSDK #2019: Where can I access camera_action on the Pixhawk 6C?

ryanpinney, 2023-04-07:
I am using mavsdk python to make missionItems, and at each point on the mission there is an option for camera_action. I will be using camera action(TAKE_PHOTO) and camera action(NONE). What pins can I connect to on the Pixhawk 6C to get these camera triggers? There is no option "actuator" setup in qgroundcontrol for me I am assuming it is because of the airframe type, so I do not know how to configure the pins.

JonasVautherin, 2023-04-07:
I think that this is more a question for PX4. The autopilot is the one "executing" the mission and actually triggering the camera in that case :+1:.

julianoes, 2023-04-10:
It's a bit confusing because there is the MAVSDK Action TAKE_PHOTO which sends the command [MAV_CMD_IMAGE_START_CAPTURE](https://mavlink.io/en/messages/common.html#MAV_CMD_IMAGE_START_CAPTURE) in PX4 to a potential camera.

However, this doesn't actually set up triggering on PX4. To use triggering, you have to set up the trigger pin in the Actuator setting, and then use MissionRaw and use the command [MAV_CMD_DO_SET_CAM_TRIGG_DIST](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_CAM_TRIGG_DIST), or [MAV_CMD_DO_SET_CAM_TRIGG_INTERVAL](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_CAM_TRIGG_INTERVAL)

julianoes, 2023-04-11:
I don't know. This seems to come up again and again and I just don't seem to get a straight answer.

jean-dupond, 2023-05-12:
> @JonasVautherin @julianoes This is probably a PX4 question too, but there is no "Actuators" menu in my QGC, is that because I selected the wrong or an incompatible airframe?

I could solve this just going to the airframe file in /init.d/airframes, disable the mixer and enabling the SYS_CTRL_ALLOC, and then setting all the necessary PWM functions up. In my case, i set `param set-default PWM_AUX_FUNC5 2000  # Camera trigger` for the pwm trigger

jean-dupond, 2023-05-12:
Then, to trigger, make sure you set up the parameters for TRIG_PWM_NEUTRAL and TRIG_PWM_SHOOT. also you could set the TRIG_INTERFACE to generic pwm and TRIG_MODE to 1 for example.

To activate the trigger just send a pymavlink command long [MAV_CMD_DO_SET_CAM_TRIGG_DIST](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_CAM_TRIGG_DIST) and wait for ACK

julianoes, 2023-05-14:
Awesome, thanks for documenting this. So we can close this? I assume it will get a bit more straightforward once v1.14 is out and the actuator config in QGC shows up by default.

closed 2024-05-27 by julianoes