mavlink/MAVSDK #2022: Serial connection error in UWP C++/CX

kousheekc, 2023-04-16:
Hello, been working to resolve this for some time now, but I am completely blocked. I am trying to create a custom ground control app in UWP with C++/CX but I cannot connect to the pixhawk via serial. I have successfully linked the mavsdk library with the app but when I run 
`mavsdk::ConnectionResult connection_result = m_Mavsdk.add_any_connection("serial://COM7");`

in the constructor of the MainPage, I get a ConnectionError result.

I have been able to create a separate C++ console app and was able to successfully connect to my pixhawk.

Thank you so much for your help.

julianoes, 2023-04-17:
Which ConnectionError result? I assume the parse error? I think you forgot the baudrate, e.g.:

` mavsdk::ConnectionResult connection_result = m_Mavsdk.add_any_connection("serial://COM7:57600");`

kousheekc, 2023-04-17:
Hello thanks for the reply, unfortunately adding the baudrate did not solve the problem. I believe according to the documentation, if no baudrate is specified it defaults to 57600, which is the baudrate set on my pixhawk.

Also as I mentioned earlier, this same code functions correctly when I use it in a pure C++ console app, the problem occurs when I try using it in a UWP C++/CX app. I have attached below the context in which I am using it: it is in the constructor of the MainPage and I am using a switch case on the connection_result along with a message dialog box in UWP to determine which is the error. This results in a ConnectionError each time.

```
MainPage::MainPage()
{
    InitializeComponent();

    mavsdk::Mavsdk m_Mavsdk;
    mavsdk::ConnectionResult connection_result = m_Mavsdk.add_any_connection("serial://COM7:57600");

    MessageDialog^ msg;

    switch (connection_result)
    {
    case mavsdk::ConnectionResult::Success:
        msg = ref new MessageDialog("Success");
        break;
    case mavsdk::ConnectionResult::Timeout:
        msg = ref new MessageDialog("Timeout");
        break;
    case mavsdk::ConnectionResult::SocketError:
        msg = ref new MessageDialog("SocketError");
        break;
    case mavsdk::ConnectionResult::BindError:
        msg = ref new MessageDialog("BindError");
        break;
    case mavsdk::ConnectionResult::SocketConnectionError:
        msg = ref new MessageDialog("SocketConnectionError");
        break;
    case mavsdk::ConnectionResult::ConnectionError:
        msg = ref new MessageDialog("ConnectionError");
        break;
    case mavsdk::ConnectionResult::NotImplemented:
        msg = ref new MessageDialog("NotImplemented");
        break;
    case mavsdk::ConnectionResult::SystemNotConnected:
        msg = ref new MessageDialog("SystemNotConnected");
        break;
    case mavsdk::ConnectionResult::SystemBusy:
        msg = ref new MessageDialog("SystemBusy");
        break;
    case mavsdk::ConnectionResult::CommandDenied:
        msg = ref new MessageDialog("CommandDenied");
        break;
    case mavsdk::ConnectionResult::DestinationIpUnknown:
        msg = ref new MessageDialog("DestinationIpUnknown");
        break;
    case mavsdk::ConnectionResult::ConnectionsExhausted:
        msg = ref new MessageDialog("ConnectionsExhausted");
        break;
    case mavsdk::ConnectionResult::ConnectionUrlInvalid:
        msg = ref new MessageDialog("ConnectionUrlInvalid");
        break;
    case mavsdk::ConnectionResult::BaudrateUnknown:
        msg = ref new MessageDialog("BaudrateUnknown");
        break;
    default:
        break;
    }

    msg->ShowAsync();
}
```

Finally, what do you mean by the parse error? Isn't the function **add_any_connection** responsible for basically checking whether the specified URL exists?

Thank you so much once again.

JonasVautherin, 2023-04-17:
> this same code functions correctly when I use it in a pure C++ console app, the problem occurs when I try using it in a UWP C++/CX app

I am not familiar with UWP apps. Does it have access to the serial port, or is there some kind of permission system?

kousheekc, 2023-04-17:
There is a permission system, but I made sure to add serial communication as a capability. It is defined this way:

```
  <Capabilities>
    <Capability Name="internetClient" />

    <DeviceCapability Name="serialcommunication">
      <Device Id="any">
        <Function Type="name:serialPort" />
      </Device>
    </DeviceCapability>
  </Capabilities>
```

JonasVautherin, 2023-04-17:
If it works in a bare CLI app and not with UWP, I would look into that permission system more closely :innocent:

closed 2024-05-27 by julianoes